OpenTabletDriver/OpenTabletDriver #1170: Failure finding non-existent express keys

Tudi20, 2021-06-11:
## Description
For Wacom CTL-472, this error is given when the tablet itself has no express keys.

```
[Detect:Info]	Searching for tablet 'Wacom CTL-472'
[Device:Warning]	Failed to set tablet feature: 02-02
[Detect:Info]	Found tablet 'Wacom CTL-472'
[Detect:Warning]	Failed to find auxiliary device, express keys may be unavailable.
˙``

## System Information:
<!-- Please fill out this information -->
| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10; 20H2; 19042.1052
| Software Version | 0.5.3.2
| Tablet           | Wacom CTL-472

adryzz, 2021-06-11:
The warnings are not device-specific. It is expected behaviour to show the warning even on devices without express keys.

Tudi20, 2021-06-11:
Reporting that you have a perfectly working tablet without express keys, or a not-perfectly working tablet with express keys that can't be used are two different things.
Unless I'm misunderstanding the error message, and you're saying that due to the software's capabilities or misconfiguration on my system express keys couldn't be used, regardless of what tablet I would be using.

Mrcubix, 2021-06-11:
the message mean what it mean, in your case, you tablet doesn't have express key, which is why this message show up.

Tudi20, 2021-06-11:
In that case the warning should show as info or be silenced. (Especially since osu!lazer likes to show every warning in OTD as a critical error.)

Mrcubix, 2021-06-11:
Re-opening for tracking purposes since ghost mentioned the possibility of removal of aux device identifier on tablet that doesn't have aux buttons

Mayla99654, 2022-09-04:
Here the same thing happens, and the cursor is static, the tablet stops responding to inputs after this message appears.

jamesbt365, 2022-09-04:
correlation != causation. Open your own issue.

closed 2022-12-26 by X9VoiD