OpenTabletDriver/OpenTabletDriver #1172: Support Request:

CamperFate, 2021-06-12:
# Support Request

Recently, I wanted to switch to opentablet driver, but I was contacted with such a problem as not detecting a graphics tablet. How to solve this problem at the bottom attached the diagnostic file

## What I've already tried

## Diagnostic Information

[h430p.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6641854/h430p.txt)
or
{
"Name": "Huion H430P",
"DigitizerIdentifiers": [
{
"Width": 121.92,
"Height": 76.2,
"MaxX": 24384.0,
"MaxY": 15240.0,
"MaxPressure": 4095,
"ActiveReportID": {
"Start": 64,
"StartInclusive": true,
"End": null,
"EndInclusive": false
},
"VendorID": 9580,
"ProductID": 109,
"InputReportLength": 12,
"OutputReportLength": null,
"ReportParser": "OpenTabletDriver.Tablet.TabletReportParser",
"FeatureInitReport": null,
"OutputInitReport": null,
"DeviceStrings": {
"201": "HUION_T176_\\d{6}$"
},
"InitializationStrings": [
200
]
},
{
"Width": 121.92,
"Height": 76.2,
"MaxX": 24384.0,
"MaxY": 15240.0,
"MaxPressure": 4095,
"ActiveReportID": {
"Start": 64,
"StartInclusive": true,
"End": null,
"EndInclusive": false
},
"VendorID": 9580,
"ProductID": 110,
"InputReportLength": 12,
"OutputReportLength": null,
"ReportParser": "OpenTabletDriver.Tablet.TabletReportParser",
"FeatureInitReport": null,
"OutputInitReport": null,
"DeviceStrings": {
"201": "HUION_T176_\\d{6}$"
},
"InitializationStrings": [
200
]
}
],
"AuxilaryDeviceIdentifiers": [],
"Attributes": {
"libinputoverride": "1"
}
}

Mrcubix, 2021-06-12:
do these 2 and export another diag and upload it here if the issue is still present 
[image]
https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0

CamperFate, 2021-06-13:
> do these 2 and export another diag and upload it here if the issue is still present
> [image]
> https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0

thanks for helping, I just had to remove the huoion drivers

closed 2021-06-13 by CamperFate