OpenTabletDriver/OpenTabletDriver #1174: Not starting on Apple Silicon mac

Knallli, 2021-06-12:
## Description
Program won't start

## System Information:
| ---------------- | ----- |
| MacOS BigSur and Monterey |
| 11.3.2/12.0 Beta |

The only sign of live shows in launchd.log:
[launchd.log](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6642056/launchd.log)

InfinityGhost, 2021-06-12:
We do not have an ARM build of OpenTabletDriver, which is why it doesn't start. I will look into building for macOS ARM in future releases.

Knallli, 2021-06-12:
I wonder why it won't run on rosetta2 though? or is that specific to just arm macs

InfinityGhost, 2021-06-12:
Might have to do with .NET 5 itself, make sure that's installed at least. I personally don't own an ARM mac so I only can speculate.

Knallli, 2021-06-12:
Okay I had the dotnet 6 preview installed. When uninstalling this and just having the dotnet 5 sdk it works

InfinityGhost, 2021-06-12:
Appears resolved then, closing.

closed 2021-06-12 by InfinityGhost

sheerun, 2021-11-29:
Aren't then plans to make it work with dotnet 6 on macos? :)

jamesbt365, 2021-11-29:
#1441

RoyiAvital, 2023-08-06:
> We do not have an ARM build of OpenTabletDriver, which is why it doesn't start. I will look into building for macOS ARM in future releases.

It would be great to have Apple M CPU support (Apple ARM / Apple Silicone).