OpenTabletDriver/OpenTabletDriver #1179: Daemon crashes at launch on fedora 34 with System.DllNotFoundException: Unable to load shared library 'libX11'

fuomag9, 2021-06-13:
## Description
Running the daemon from the latest precompiled binary will result in a crash on fedora 34

Here's the strack trace
```
sudo ./OpenTabletDriver.Daemon 
[Detect:Info]   Searching for tablet 'VEIKK A30'
[Detect:Info]   Searching for tablet 'VEIKK S640'
[Detect:Info]   Searching for tablet 'VEIKK A50'
[Detect:Info]   Searching for tablet 'VEIKK A15'
[Detect:Info]   Searching for tablet 'VEIKK A15 Pro'
[Detect:Info]   Searching for tablet 'XP-Pen Star G640S'
[Detect:Info]   Searching for tablet 'XP-Pen Deco 01 V2'
[Detect:Info]   Searching for tablet 'XP-Pen Deco Pro Small'
[Detect:Info]   Searching for tablet 'XP-Pen Deco 01'
[Detect:Info]   Searching for tablet 'XP-Pen Star 03'
[Detect:Info]   Searching for tablet 'XP-Pen Deco 02'
[Detect:Info]   Searching for tablet 'XP-Pen Deco 03'
[Detect:Info]   Searching for tablet 'XP-Pen Star G640 V2'
[Detect:Info]   Searching for tablet 'XP-Pen Star G960S Plus'
[Detect:Info]   Searching for tablet 'XP-Pen Deco Mini7'
[Detect:Info]   Searching for tablet 'XP-Pen Star 06C'
[Detect:Info]   Searching for tablet 'XP-Pen Star G540'
[Detect:Info]   Searching for tablet 'XP-Pen Star G430S'
[Detect:Info]   Searching for tablet 'XP-Pen Star G430'
[Detect:Info]   Searching for tablet 'XP-Pen Artist 12 Pro'
[Detect:Info]   Searching for tablet 'XP-Pen Star G540 Pro'
[Detect:Info]   Searching for tablet 'XP-Pen Star G960'
[Detect:Info]   Searching for tablet 'XP-Pen Star 06'
[Detect:Info]   Searching for tablet 'XP-Pen Star G640'
[Detect:Debug]  Using device 'STAR G640'.
[Detect:Debug]  Using report parser type 'OpenTabletDriver.Tablet.TabletReportParser'.
[Detect:Debug]  Device path: /sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1:1.2/0003:28BD:0094.000D/hidraw/hidraw3
[Device:Debug]  Set tablet output: 02-B0-02
[Detect:Info]   Found tablet 'XP-Pen Star G640'
Unhandled exception. System.DllNotFoundException: Unable to load shared library 'libX11' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibX11: cannot open shared object file: No such file or directory
   at OpenTabletDriver.Native.Linux.Xorg.XLib.sys_XOpenDisplay(Char* display)
   at OpenTabletDriver.Native.Linux.Xorg.XLib.XOpenDisplay(Char* display)
   at OpenTabletDriver.Desktop.Interop.Display.XScreen..ctor()
   at OpenTabletDriver.Desktop.Interop.SystemInterop.ConstructLinuxDisplay()
   at OpenTabletDriver.Desktop.Interop.SystemInterop.<>c.<.cctor>b__18_3()
   at System.Lazy`1.ViaFactory(LazyThreadSafetyMode mode) in /_/src/libraries/System.Private.CoreLib/src/System/Lazy.cs:line 321
   at System.Lazy`1.ExecutionAndPublication(LazyHelper executionAndPublication, Boolean useDefaultConstructor) in /_/src/libraries/System.Private.CoreLib/src/System/Lazy.cs:line 345
   at System.Lazy`1.CreateValue() in /_/src/libraries/System.Private.CoreLib/src/System/Lazy.cs:line 425
   at System.Lazy`1.get_Value() in /_/src/libraries/System.Private.CoreLib/src/System/Lazy.cs:line 505
   at OpenTabletDriver.Desktop.Interop.SystemInterop.get_VirtualScreen()
   at OpenTabletDriver.Desktop.Settings.get_Default()
   at OpenTabletDriver.Daemon.DriverDaemon.ResetSettings()
   at OpenTabletDriver.Daemon.DriverDaemon.LoadUserSettings()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_1(Object state) in /_/src/libraries/System.Private.CoreLib/src/System/Threading/Tasks/Task.cs:line 1898
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute() in /_/src/libraries/System.Private.CoreLib/src/System/Threading/ThreadPool.cs:line 853
   at System.Threading.ThreadPoolWorkQueue.Dispatch() in /_/src/libraries/System.Private.CoreLib/src/System/Threading/ThreadPool.cs:line 641
   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback() in /_/src/coreclr/src/System.Private.CoreLib/src/System/Threading/ThreadPool.CoreCLR.cs:line 29
Aborted
```

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | fedora 34
| Software Version | 0.5.3.2
| Tablet          | XP-Pen Star G640

gonX, 2021-06-13:
Since you're running the command with `sudo` it's likely that `WAYLAND_DISPLAY` is getting filtered by it and therefore is causing that issue. Try running it without `sudo`.

Wayland detection is currently not very solid in OTD, but it should work as long as you have the environment variable `WAYLAND_DISPLAY` set.

If it's not set, it will fall back to an X11 session which Fedora 34 dropped in favor of Wayland AFAIK.

fuomag9, 2021-06-13:
> Since you're running the command with `sudo` it's likely that `WAYLAND_DISPLAY` is getting filtered by it and therefore is causing that issue. Try running it without `sudo`.
>
> Wayland detection is [...]

Unfortunately the same error happens without sudo as well (I still need to fix the `DeviceUnathorizedAccessException:Error` that happens if I don't use sudo). At the moment I'm using X11 as my session, since wayland doesn't work very well for me (fedora 34 included both wayland and x11 by default btw)

gonX, 2021-06-13:
If you're using X11, make sure libX11 is installed

There's a FAQ topic on missing permissions on Linux. You need to install the udev rules as described [HERE](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception)

fuomag9, 2021-06-13:
`sudo dnf install libX11-devel libXrandr-devel` successfully fixed the libx11 issue, thank you :)

closed 2021-06-13 by fuomag9