OpenTabletDriver/OpenTabletDriver #1180: Support Request: can't install the installer

fouzo53, 2021-06-13:
# Support Request

## What I've already tried

i have no clue why its cant install even the portable, i haved install .net 5.0
## Diagnostic Information

X9VoiD, 2021-06-13:
Install .NET 5.0, and installer installs OTD, it doesn't install itself, so no, installer will not install itself.

closed 2021-06-13 by X9VoiD

fouzo53, 2021-06-13:
which OTD i have download? office deployment tool?

fouzo53, 2021-06-13:
which OTD i have download? Oracle Traffic Director?

fouzo53, 2021-06-13:
i haved download .net, but the OTD installer just wont pop up

X9VoiD, 2021-06-13:
.NET 5.0 is not OTD itself, it's a runtime required by OTD to properly run. OTD portable can be found in [Releases](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.5.3.2). ([direct link](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/download/v0.5.3.2/OpenTabletDriver.win-x64.zip))

Installer/Updater is deprecated and is not the recommended way anymore to setup OTD.

fouzo53, 2021-06-13:
i mean... OTD don't open when i have clicked