OpenTabletDriver/OpenTabletDriver #1181: Support Request: Pen Pressure doesn't work on Huion H640p on Sai

RelicSpoon, 2021-06-13:
# Support Request
My Pen pressure doesn't work on Paint Tool Sai 
After switching to OTD because Devocub doesn't work on new Windows Update. Before OTD. I use Devocub which my pen pressure works very well with Windows Ink. Now it doesn't work. I have download VMulti and Windows Ink to see if it works but no luck. 
## What I've already tried

## Diagnostic Information
 {
  "App Version": "OpenTabletDriver v0.5.3.1",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19042.0",
    "VersionString": "Microsoft Windows NT 10.0.19042.0"
  },
  "Environment Variables": {
    "FPS_BROWSER_APP_PROFILE_STRING": "Internet Explorer",
    "FPS_BROWSER_USER_PROFILE_STRING": "Default",
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "windir": "C:\\WINDOWS",
    "APPDATA": "C:\\Users\\surne\\AppData\\Roaming",
    "PROCESSOR_LEVEL": "6",
    "NUMBER_OF_PROCESSORS": "8",
    "Path": "C:\\Program Files (x86)\\Common Files\\Oracle\\Java\\javapath;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Program Files\\dotnet\\;C:\\Users\\surne\\AppData\\Local\\Microsoft\\WindowsApps;",
    "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC",
    "PUBLIC": "C:\\Users\\Public",
    "TMP": "C:\\Users\\surne\\AppData\\Local\\Temp",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "OneDriveConsumer": "C:\\Users\\surne\\OneDrive",
    "LOCALAPPDATA": "C:\\Users\\surne\\AppData\\Local",
    "TEMP": "C:\\Users\\surne\\AppData\\Local\\Temp",
    "ProgramFiles": "C:\\Program Files",
    "VBOX_MSI_INSTALL_PATH": "D:\\Program Files\\Oracle\\VirtualBox\\",
    "ProgramData": "C:\\ProgramData",
    "PROCESSOR_IDENTIFIER": "Intel64 Family 6 Model 94 Stepping 3, GenuineIntel",
    "OS": "Windows_NT",
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules",
    "COMPUTERNAME": "DESKTOP-L04AETT",
    "HOMEDRIVE": "C:",
    "HOMEPATH": "\\Users\\surne",
    "USERDOMAIN_ROAMINGPROFILE": "DESKTOP-L04AETT",
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "ProgramW6432": "C:\\Program Files",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "ComSpec": "C:\\WINDOWS\\system32\\cmd.exe",
    "SystemRoot": "C:\\WINDOWS",
    "SESSIONNAME": "Console",
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "USERNAME": "surne",
    "USERPROFILE": "C:\\Users\\surne",
    "OneDrive": "C:\\Users\\surne\\OneDrive",
    "PROCESSOR_REVISION": "5e03",
    "SystemDrive": "C:",
    "LOGONSERVER": "\\\\DESKTOP-L04AETT",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "USERDOMAIN": "DESKTOP-L04AETT"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#hid&col03#1&4784345&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 255,
      "ProductID": 47820,
      "CanOpen": true,
      "ReleaseNumberBcd": 1,
      "ReleaseNumber": "0.1",
      "ProductVersion": 1,
      "Manufacturer": "DJP Inc.",
      "ProductName": "Virtual Multitouch Device",
      "SerialNumber": "123123123",
      "MaxInputReportLength": 10,
[... 953 lines cut ...]
      "Message": "Tip Binding: [VoiDPlugins.OutputMode.WinInkButtonHandler]@1%",
      "Level": 1
    },
    {
      "Time": "2021-06-13T19:03:58.5111434+08:00",
      "Group": "Settings",
      "Message": "Pen Bindings: [0, ], [1, ]",
      "Level": 1
    },
    {
      "Time": "2021-06-13T19:03:58.5111481+08:00",
      "Group": "Settings",
      "Message": "Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ], [6, ], [7, ]",
      "Level": 1
    },
    {
      "Time": "2021-06-13T19:03:58.5111528+08:00",
      "Group": "Settings",
      "Message": "Driver is auto-enabled.",
      "Level": 1
    }
  ]
}

[image]

SnowLire, 2021-06-13:
Vmulti is broken everywhere, and OTD's version is not an exception.
Adding on to that, Paint Tool Sai does not support Windows Ink API, meaning that pressure sensitivity cannot be "enabled" in the program regardless of vmulti working or not

RelicSpoon, 2021-06-14:
> Vmulti is broken everywhere, and OTD's version is not an exception.
> Adding on to that, Paint Tool Sai does not support Windows Ink API, meaning that pressure sensitivity cannot be "enabled" in the program regardless of vmulti working or not

So any way to fix this? Or I have to use a different art program?

SnowLire, 2021-06-14:
> So any way to fix this? Or I have to use a different art program?

Currently we're waiting on a fix for vmulti, although yes, you would need to use a different art program afterwards.

jamesbt365, 2021-12-28:
paint tool sai doesn't support winink, wintab support is tracked in #1230

closed 2021-12-28 by jamesbt365