OpenTabletDriver/OpenTabletDriver #1185: Huion H430P - No tablet detected

jhemmmm, 2021-06-14:
## Description

Hello,

I have Huion H430P. I was using the TabletDriver until it stopped working for a dual monitor after the last update my windows had.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System |
| Software Version |
| Tablet           |

Windows 10  Pro N
Version: 20H2
OS Build: 19042.1052
Software Version: Latest Version
Tablet: Huion H430P

Mrcubix, 2021-06-14:
Complete the template with a diagnostic generated in OTD before we proceed.

jhemmmm, 2021-06-14:
Here's the exported diagnostics.
```
{
  "App Version": "OpenTabletDriver v0.5.3.1",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19042.0",
    "VersionString": "Microsoft Windows NT 10.0.19042.0"
  },
  "Environment Variables": {
    "VBOX_MSI_INSTALL_PATH": "D:\\Program Files\\Oracle\\VirtualBox\\",
    "ProgramData": "C:\\ProgramData",
    "GOPATH": "C:\\Users\\warro\\go",
    "LOGONSERVER": "\\\\DESKTOP-U8UM6LE",
    "LOCALAPPDATA": "C:\\Users\\warro\\AppData\\Local",
    "USERPROFILE": "C:\\Users\\warro",
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules;C:\\Program Files\\Intel\\Wired Networking\\",
    "SystemRoot": "C:\\WINDOWS",
    "VS120COMNTOOLS": "D:\\Program Files (x86)\\Microsoft Visual Studio 12.0\\Common7\\Tools\\",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "OneDriveConsumer": "C:\\Users\\warro\\OneDrive",
    "APPDATA": "C:\\Users\\warro\\AppData\\Roaming",
    "PUBLIC": "C:\\Users\\Public",
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "Bento4": "C:\\Program Files (x86)\\Bento4",
    "HOMEPATH": "\\Users\\warro",
    "PROCESSOR_LEVEL": "23",
    "USERNAME": "warro",
    "VBOX_HWVIRTEX_IGNORE_SVM_IN_USE": "1",
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "AMDRMSDKPATH": "C:\\Program Files\\AMD\\RyzenMasterSDK\\",
    "NUMBER_OF_PROCESSORS": "16",
    "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC",
    "ProgramFiles": "C:\\Program Files",
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "COMPUTERNAME": "DESKTOP-U8UM6LE",
    "VS140COMNTOOLS": "C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\Common7\\Tools\\",
    "windir": "C:\\WINDOWS",
    "VS110COMNTOOLS": "C:\\Program Files (x86)\\Microsoft Visual Studio 11.0\\Common7\\Tools\\",
    "OneDrive": "C:\\Users\\warro\\OneDrive",
    "ProgramW6432": "C:\\Program Files",
    "USERDOMAIN_ROAMINGPROFILE": "DESKTOP-U8UM6LE",
    "Path": "C:\\Program Files (x86)\\Common Files\\Oracle\\Java\\javapath;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Program Files\\NVIDIA Corporation\\NVIDIA NvDLISR;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\dotnet\\;C:\\Program Files\\Microsoft SQL Server\\130\\Tools\\Binn\\;C:\\Program Files\\Common Files\\Autodesk Shared\\;C:\\Program Files (x86)\\Autodesk\\Backburner\\;D:\\Program Files\\Git\\cmd;C:\\Program Files\\Microsoft SQL Server\\110\\Tools\\Binn\\;C:\\Program Files (x86)\\Microsoft SDKs\\TypeScript\\1.0\\;C:\\Users\\warro\\AppData\\Roaming\\Composer\\vendor\\bin;C:\\Program Files\\FFMPEG;D:\\Program Files\\HashiCorp\\Vagrant\\bin;C:\\Users\\warro\\AppData\\Local\\Programs\\Microsoft VS Code\\bin;D:\\Go\\bin;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Users\\warro\\scoop\\shims;C:\\Users\\warro\\AppData\\Local\\Yarn\\config\\global\\node_modules\\.bin;C:\\Users\\warro\\AppData\\Roaming\\Composer\\vendor\\bin;C:\\Users\\warro\\AppData\\Roaming\\npm;D:\\laragon\\bin;D:\\laragon\\bin\\apache\\httpd-2.4.35-win64-VC15\\bin;D:\\laragon\\bin\\composer;D:\\laragon\\bin\\git\\bin;D:\\laragon\\bin\\git\\cmd;D:\\laragon\\bin\\git\\mingw64\\bin;D:\\laragon\\bin\\git\\usr\\bin;D:\\laragon\\bin\\laragon\\utils;D:\\laragon\\bin\\mysql\\mysql-5.7.24-winx64\\bin;D:\\laragon\\bin\\nginx\\nginx-1.16.0;D:\\laragon\\bin\\ngrok;D:\\laragon\\bin\\nodejs\\node-v12;D:\\laragon\\bin\\notepad++;D:\\laragon\\bin\\php\\php-7.2.19-Win32-VC15-x64;D:\\laragon\\bin\\putty;D:\\laragon\\bin\\python\\python-3.6.1;D:\\laragon\\bin\\python\\python-3.6.1\\Scripts;D:\\laragon\\bin\\redis\\redis-x64-3.2.100;D:\\laragon\\bin\\telnet;D:\\laragon\\bin\\yarn\\bin;D:\\laragon\\usr\\bin;C:\\Users\\warro\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Users\\warro\\AppData\\Local\\Programs\\Microsoft VS Code\\bin;C:\\Users\\warro\\AppData\\Local\\GitHubDesktop\\bin;C:\\Users\\warro\\AppData\\Local\\atom\\bin;C:\\Users\\warro\\go\\bin;C:\\Users\\warro\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Program Files (x86)\\Bento4;;D:\\Program Files (x86)\\Fiddler",
    "OS": "Windows_NT",
    "ADSK_CLM_WPAD_PROXY_CHECK": "FALSE",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "USERDOMAIN": "DESKTOP-U8UM6LE",
    "SESSIONNAME": "Console",
    "JD2_HOME": "D:\\Program Files\\JDownloader 2.0",
    "SystemDrive": "C:",
    "TEMP": "C:\\Users\\warro\\AppData\\Local\\Temp",
    "PROCESSOR_IDENTIFIER": "AMD64 Family 23 Model 8 Stepping 2, AuthenticAMD",
    "HOMEDRIVE": "C:",
    "ComSpec": "C:\\WINDOWS\\system32\\cmd.exe",
    "TMP": "C:\\Users\\warro\\AppData\\Local\\Temp",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "PROCESSOR_REVISION": "0802",
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "ADSK_3DSMAX_x64_2017": "D:\\Program Files\\Autodesk\\3ds Max 2017\\"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_0518&pid_5111&mi_01&col04#8&1f5ba818&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 1304,
      "ProductID": 20753,
      "CanOpen": true,
      "ReleaseNumberBcd": 512,
      "ReleaseNumber": "2.0",
      "ProductVersion": 512,
[... 5229 lines cut ...]
      "Message": "Searching for tablet 'XP-Pen Star G640S'",
      "Level": 1
    },
    {
      "Time": "2021-06-15T05:25:41.8681194+08:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'XP-Pen Star G960'",
      "Level": 1
    },
    {
      "Time": "2021-06-15T05:25:41.8681269+08:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'XP-Pen Star G960S Plus'",
      "Level": 1
    },
    {
      "Time": "2021-06-15T05:25:41.8681348+08:00",
      "Group": "Detect",
      "Message": "No tablet found.",
      "Level": 1
    }
  ]
}

```

zooxs, 2021-08-28:
i have some issue but in manjaro linux, what should i do?

Mrcubix, 2021-08-28:
make your on issue since it will be different

Mrcubix, 2021-08-28:
> Here's the exported diagnostics.
> 
> Diagnostics
> {
>   "App Version": "OpenTabletDriver v0.5.3.1",
>   "Operating System": {
>     "Platform": 2,
>     "ServicePack": "",
>     "Version": "10.0.19042.0",
>     "VersionString": "Microsoft Windows NT 10.0.19042.0"
>   },
>   "Environment Variables": {
>     "VBOX_MSI_INSTALL_PATH": "D:\\Program Files\\Oracle\\VirtualBox\\",
>     "ProgramData": "C:\\ProgramData",
>     "GOPATH": "C:\\Users\\warro\\go",
>     "LOGONSERVER": "\\\\DESKTOP-U8UM6LE",
>     "LOCALAPPDATA": "C:\\Users\\warro\\AppData\\Local",
>     "USERPROFILE": "C:\\Users\\warro",
>     "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules;C:\\Program Files\\Intel\\Wired Networking\\",
>     "SystemRoot": "C:\\WINDOWS",
>     "VS120COMNTOOLS": "D:\\Program Files (x86)\\Microsoft Visual Studio 12.0\\Common7\\Tools\\",
>     "CommonProgramW6432": "C:\\Program Files\\Common Files",
>     "OneDriveConsumer": "C:\\Users\\warro\\OneDrive",
>     "APPDATA": "C:\\Users\\warro\\AppData\\Roaming",
>     "PUBLIC": "C:\\Users\\Public",
>     "CommonProgramFiles": "C:\\Program Files\\Common Files",
>     "Bento4": "C:\\Program Files (x86)\\Bento4",
>     "HOMEPATH": "\\Users\\warro",
>     "PROCESSOR_LEVEL": "23",
>     "USERNAME": "warro",
>     "VBOX_HWVIRTEX_IGNORE_SVM_IN_USE": "1",
>     "ProgramFiles(x86)": "C:\\Program Files (x86)",
>     "AMDRMSDKPATH": "C:\\Program Files\\AMD\\RyzenMasterSDK\\",
>     "NUMBER_OF_PROCESSORS": "16",
>     "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC",
>     "ProgramFiles": "C:\\Program Files",
>     "ALLUSERSPROFILE": "C:\\ProgramData",
>     "COMPUTERNAME": "DESKTOP-U8UM6LE",
>     "VS140COMNTOOLS": "C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\Common7\\Tools\\",
>     "windir": "C:\\WINDOWS",
>     "VS110COMNTOOLS": "C:\\Program Files (x86)\\Microsoft Visual Studio 11.0\\Common7\\Tools\\",
>     "OneDrive": "C:\\Users\\warro\\OneDrive",
>     "ProgramW6432": "C:\\Program Files",
>     "USERDOMAIN_ROAMINGPROFILE": "DESKTOP-U8UM6LE",
>     "Path": "C:\\Program Files (x86)\\Common Files\\Oracle\\Java\\javapath;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Program Files\\NVIDIA Corporation\\NVIDIA NvDLISR;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\dotnet\\;C:\\Program Files\\Microsoft SQL Server\\130\\Tools\\Binn\\;C:\\Program Files\\Common Files\\Autodesk Shared\\;C:\\Program Files (x86)\\Autodesk\\Backburner\\;D:\\Program Files\\Git\\cmd;C:\\Program Files\\Microsoft SQL Server\\110\\Tools\\Binn\\;C:\\Program Files (x86)\\Microsoft SDKs\\TypeScript\\1.0\\;C:\\Users\\warro\\AppData\\Roaming\\Composer\\vendor\\bin;C:\\Program Files\\FFMPEG;D:\\Program Files\\HashiCorp\\Vagrant\\bin;C:\\Users\\warro\\AppData\\Local\\Programs\\Microsoft VS Code\\bin;D:\\Go\\bin;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Users\\warro\\scoop\\shims;C:\\Users\\warro\\AppData\\Local\\Yarn\\config\\global\\node_modules\\.bin;C:\\Users\\warro\\AppData\\Roaming\\Composer\\vendor\\bin;C:\\Users\\warro\\AppData\\Roaming\\npm;D:\\laragon\\bin;D:\\laragon\\bin\\apache\\httpd-2.4.35-win64-VC15\\bin;D:\\laragon\\bin\\composer;D:\\laragon\\bin\\git\\bin;D:\\laragon\\bin\\git\\cmd;D:\\laragon\\bin\\git\\mingw64\\bin;D:\\laragon\\bin\\git\\usr\\bin;D:\\laragon\\bin\\laragon\\utils;D:\\laragon\\bin\\mysql\\mysql-5.7.24-winx64\\bin;D:\\laragon\\bin\\nginx\\nginx-1.16.0;D:\\laragon\\bin\\ngrok;D:\\laragon\\bin\\nodejs\\node-v12;D:\\laragon\\bin\\notepad++;D:\\laragon\\bin\\php\\php-7.2.19-Win32-VC15-x64;D:\\laragon\\bin\\putty;D:\\laragon\\bin\\python\\python-3.6.1;D:\\laragon\\bin\\python\\python-3.6.1\\Scripts;D:\\laragon\\bin\\redis\\redis-x64-3.2.100;D:\\laragon\\bin\\telnet;D:\\laragon\\bin\\yarn\\bin;D:\\laragon\\usr\\bin;C:\\Users\\warro\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Users\\warro\\AppData\\Local\\Programs\\Microsoft VS Code\\bin;C:\\Users\\warro\\AppData\\Local\\GitHubDesktop\\bin;C:\\Users\\warro\\AppData\\Local\\atom\\bin;C:\\Users\\warro\\go\\bin;C:\\Users\\warro\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Program Files (x86)\\Bento4;;D:\\Program Files (x86)\\Fiddler",
>     "OS": "Windows_NT",
>     "ADSK_CLM_WPAD_PROXY_CHECK": "FALSE",
>     "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
>     "USERDOMAIN": "DESKTOP-U8UM6LE",
>     "SESSIONNAME": "Console",
>     "JD2_HOME": "D:\\Program Files\\JDownloader 2.0",
>     "SystemDrive": "C:",
>     "TEMP": "C:\\Users\\warro\\AppData\\Local\\Temp",
>     "PROCESSOR_IDENTIFIER": "AMD64 Family 23 Model 8 Stepping 2, AuthenticAMD",
>     "HOMEDRIVE": "C:",
>     "ComSpec": "C:\\WINDOWS\\system32\\cmd.exe",
>     "TMP": "C:\\Users\\warro\\AppData\\Local\\Temp",
>     "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
>     "PROCESSOR_REVISION": "0802",
>     "PROCESSOR_ARCHITECTURE": "AMD64",
>     "ADSK_3DSMAX_x64_2017": "D:\\Program Files\\Autodesk\\3ds Max 2017\\"
>   },
>   "HID Devices": [
>     {
>       "DevicePath": "\\\\?\\hid#vid_0518&pid_5111&mi_01&col04#8&1f5ba818&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}",
>       "VendorID": 1304,
>       "ProductID": 20753,
>       "CanOpen": true,
>       "ReleaseNumberBcd": 512,
>       "ReleaseNumber": "2.0",
[... cut ...]
>       "Group": "Detect",
>       "Message": "Searching for tablet 'XP-Pen Star G960'",
>       "Level": 1
>     },
>     {
>       "Time": "2021-06-15T05:25:41.8681269+08:00",
>       "Group": "Detect",
>       "Message": "Searching for tablet 'XP-Pen Star G960S Plus'",
>       "Level": 1
>     },
>     {
>       "Time": "2021-06-15T05:25:41.8681348+08:00",
>       "Group": "Detect",
>       "Message": "No tablet found.",
>       "Level": 1
>     }
>   ]
> }

https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0

zooxs, 2021-08-28:
> 
> 
> make your on issue since it will be different

ok, thx for that

jhemmmm, 2021-09-13:
Fixed with the new update. Thanks.

closed 2021-09-13 by jhemmmm