OpenTabletDriver/OpenTabletDriver #1188: Support Request: Tablet Area doesn't reach the whole screen

lex1k1337, 2021-06-15:
I have two monitors and after the latest update of windows my mapping was broken. In Hawku Drivers, when i set only one monitor it's mapped only half of this one. After this case i tried to use OTD and in OTD i have another issue. I can't choose the first monitor area. I tried to drag area to the left, but it's have invisible barrier. Any one know how to fix it?
[xp-pen g430.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6654397/xp-pen.g430.zip)
[image]

AbstractQbit, 2021-06-15:
Right click display area > set to display > select the needed display.

lex1k1337, 2021-06-15:
I can't click on need display because after i hover on set to display and drag a cursor the menu is disappearing

lex1k1337, 2021-06-15:
Ok, with the TAB i do it. Thx for help

X9VoiD, 2021-06-15:
This is already fixed on latest master, the next major release will contain the fix.

closed 2021-06-15 by X9VoiD