OpenTabletDriver/OpenTabletDriver #1194: help me plz

XXX6000USER, 2021-06-16:
## Description

well my driver is bugged when i save or apply it delete the tablet then i press ctrl+d to detect the tablet again 
and when programming the keys on tablet or pen for mouse click it does not work how ever it works with any other key 
except as mouse right or left clikc the only thing works as left click mouse is the pen tip only 
## System Information:
 
| Name             | Value |
| ---------------- | ----- |
| Operating System | arch linux
| Software Version | latest
| Tablet           | huion h430p and huion hs64

SnowLire, 2021-06-16:
Do help > export diagnostic in OTD, zip the created file and post it here.

gonX, 2021-07-09:
Barrel buttons not working should be fixed with #1233 being merged

Please specify your software version instead of latest - this could've been latest when you installed the package X months ago! The version specified in the title of OTD's GUI is a good start.

I suggest using the latest _release_ (latest tagged version) instead of using the latest master - on Arch that's the `opentabletdriver` package.

InfinityGhost, 2021-07-18:
Closing as #1233 has been merged.

closed 2021-07-18 by InfinityGhost