OpenTabletDriver/OpenTabletDriver #1195: Support Request: Problems with CTL-472

Akamezzu, 2021-06-16:
# Support Request
First of all Hello, 
I rly like ur driver and had no problems with it since.
Im using a CTL-472.
Since the windows update i made yesterday (21H1) the driver does work, but 
it doesnt rly.
I always played with the same area and configs and made no changes yet.
But since the update it feels like my area doesnt work anymore bc when i drag Horizontal im slowed, and when i drag vertical its way faster, but its not like i made area changes.

## What I've already tried
I already tried this with other drivers (devoclub ... ) and its in every driver the same problem.
But since this is my main and favorite driver im hoping u could probably help me with my problem

## Diagnostic Information
I tried uninstalling everything and install again idk

X9VoiD, 2021-06-16:
All VMulti-based tablet drivers, including OTD's VMulti-based plugins are broken and there are no fixes for it. Windows 10 21H1 is confining the whole virtual desktop space to just the primary monitor, and fixing it is impossible.

Just use OTD's normal input modes, they are better, and have lower latency (tho it's in the microsecond scale and no human being is capable of perceiving that).

Akamezzu, 2021-06-16:
> All VMulti-based tablet drivers, including OTD's VMulti-based plugins are broken and there are no fixes for it. Windows 10 21H1 is confining the whole virtual desktop space to just the primary [...]

Akamezzu, 2021-06-16:
i ll try it

closed 2021-06-16 by Akamezzu

Akamezzu, 2021-06-16:
i uninstalled vmulti drivers and its still the same problem idk