OpenTabletDriver/OpenTabletDriver #1197: Add support for UC-Logic PF1209

peterfab9845, 2021-06-17:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | UC-Logic |
| Product Name    | PF1209 |
| Width           | 305 mm |
| Height (active)          | 229 mm |
| Pressure Levels | 1024 |
| LPI             | 2000 |

AKA Monoprice #6815

I've figured out a configuration (attached) which works for this tablet, but I think it might need some more work that I'm not sure how to do.
(I think the tablet is supposed to have a strip of "buttons" at the top, but the cover is long gone. There's a ~10mm strip that all registers as the very top of the active area, but I don't see any difference in the reports by tapping in that area.)
[PF1209.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6668475/PF1209.json.txt)
[90-opentabletdriver_PF1209.rules.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6668480/90-opentabletdriver_PF1209.rules.txt)

## Diagnostic Information

[diagnostics.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6668465/diagnostics.json.txt)

InfinityGhost, 2021-06-17:
> I think the tablet is supposed to have a strip of "buttons" at the top, but the cover is long gone. There's a ~10mm strip that all registers as the very top of the active area, but I don't see any difference in the reports by tapping in that area.

We do not support soft buttons in any way, so this is to be expected. It will just be treated as part of the normal tablet area.

Can you run a string dump on this tablet? Open the device string reader, type in the VID and PID and dump all strings.

InfinityGhost, 2021-06-17:
Went ahead and converted your uploaded configuration to the v0.6 specification.

[PF1209.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6670405/PF1209.json.txt)

peterfab9845, 2021-06-17:
Sure, here it is:
[strings.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6673384/strings.txt)

I also found that reading some particular strings did things:
104 - flip coordinates in X
105 - flip coordinates in Y
106 - reset 104, 105, 109
109 - double the LPI! (It does seem to be real - I recorded a bunch of X positions, and I see consecutive numbers, so it's not just doubling the coordinates. New max X = 47999, max Y = 35999)
104, 105, and 109 can be used in any combination.

Another thing I noticed (might be fixed in the current git version, I'm on 0.5.3.2) is that plugging in the tablet with OTD running gets me the DeviceUnathorizedAccessException, but then re-detecting afterwards works fine. The wiki fix that the error mentions doesn't seem to change anything.

InfinityGhost, 2021-06-17:
> Finally, regarding the soft button area - any reports in that 10mm strip show up with a Y coordinate of 0, so I didn't consider it part of the active area. Including this strip, the total height is 239 instead of 229 mm, but I'm not sure how you'd want to deal with this.

We would use 229mm considering that it doesn't report any changes in the vertical axis on the strip, otherwise the aspect ratio and scale would be incorrectly handled.

peterfab9845, 2021-06-18:
Is it fairly well defined what has to happen to integrate a new tablet?
I could try my hand at a PR if I know what needs to be done besides adding the configuration.

InfinityGhost, 2021-06-18:
All that has to be done is adding the tablet to TABLETS.md in the pr and it would be done.

PR #1204 (2021-06-20): Add support for UC-Logic PF1209
Just basic support - binding for tip contact while using the buttons is possible, but not included in this PR since it depends on >2 pen buttons.

Partially resolves #1197
closed 2021-06-20 by InfinityGhost