OpenTabletDriver/OpenTabletDriver #1198: Support Request: tablet wont work at all in the debugger i see the cursor move but i cant get my cursor to move on desktop

ahdorable, 2021-06-18:
# Support Request
i would like to able to use my tablet again it wont do anything after ive reinstalled the software and tried all of the fixes ive seen on the windows faq

## What I've already tried

## Diagnostic Information

vsimochi, 2021-06-18:
also happening to me too

SnowLire, 2021-06-18:
Fill out the support issue template correctly first, then we can continue.

Lorphex, 2021-07-17:
Happening to me after updating to v0.5.3.3
I've tried switching nubs, restarting OTD, restarting my PC, etc.
Here's my diagnostic:
[OTDissue1207.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6834571/OTDissue1207.zip)

Mrcubix, 2021-07-17:
Open your own issue

InfinityGhost, 2021-10-30:
Stale support request, assuming it has been resolved. This can be reopened in the case this was a mistake.

closed 2021-10-30 by InfinityGhost