OpenTabletDriver/OpenTabletDriver #1201: Support Request: an different version UGEE M708 (may XP-PEN M708)

waitlamp, 2021-06-19:
hello, 
I have a newly bought UGEE M708
I noticed in issue #430 
there is an configuration file

however my M708 have entirely different Vendor ID and Product ID

Due to Corporate acquisitions and restructuring, XP-PEN, UGEE now are actually the same corporation.

my M708 Vendor ID is `10429`, same as XP-PEN. Product ID is `2311` which is close to XP-PEN Deco 01 V2 `2309`
you may have noticed that XP-PEN devices Manufacturer is `UGTABLET`, is approved they are same.

After copy an Deco 01 v2 configuration file, and modified DeviceString 
`      "DeviceStrings": {
        "4": "UG902_BPG1006"
      }.` 

my tablet can be detected.
[image]

The work area has a little problem, use default configuration, cursor can't move to the bottom of screen, like
[image]

and the debugger
[image]

so how should I adjust configuration file to support my M708 further, make work area correct?

and how to test if tilt is work?

is it possible to support this new version M708?

Diagnostics file
[UGEE_M708_new_diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6679993/UGEE_M708_new_diag.txt)

And my configuration file (edited DeviceString and name from XP-PEN deco 01 v2)

[M708XP-PEN_conf.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6679992/M708XP-PEN_conf.json.txt)

X9VoiD, 2021-06-19:
Try to get the highest possible value for Position in tablet debugger (should be lower right of the tablet when oriented upright), then enter here the values that you got.

waitlamp, 2021-06-19:
[50800,30480]
I tried to edit configuration file `DigitizerIdentifiers:`  `MAXY` value from `31750` to `30480`, work area now correct.
Do I need to change anything else in the configuration?

waitlamp, 2021-06-19:
I found it seems has no pressure sensitive in other software 
eg. krita, OneNote
but debugger can identify the pressure from 0 to 8191

waitlamp, 2021-06-19:
[image]
and windows ink plugin occur an error

waitlamp, 2021-06-19:
it seems Tablet buttons, tilt not work
but pressure worked well with windows ink plugin.

I 'm not sure about how to test if tilt worked correctly

so may be it can be add to supported tablets?
and there is a difference in #430 
so how to deal with two different version M708 ?

waitlamp, 2021-06-20:
#1203

closed 2021-06-20 by waitlamp