OpenTabletDriver/OpenTabletDriver #1205: Support I2C HID devices?

waitlamp, 2021-06-20:
Some laptop or 2-in-1 devices also equipped with digitizer
Tablet or graphic tablet normally use USB port
but these tablet PCs usually have I2C HID digitizer devices. 

(usually, they were detected as more than one devices in device manger)

Do you have any plans to adapt to I2C devices in the future?

thanks.

adryzz, 2021-06-20:
Yes, but we need a way to access such I2C devices from user-level .NET/native C++ code

InfinityGhost, 2021-06-20:
I2C HID will be supported, but not anytime soon.

gonX, 2021-12-01:
Related to #1390 as its another request for a different backend

Twilight-lxy, 2025-06-04:
I have a Lenovo Duet 2020 tablet, which is also a wocam I2C device. I want to use a stylus in Ubuntu, but Lenovo does not provide a driver for Ubuntu. I also hope to receive support through the OTD software.