OpenTabletDriver/OpenTabletDriver #1207: Support Request: Wacom CTL-4100 pen movement not detected

Lorphex, 2021-06-21:
# Support Request
Moving the pen does not move the cursor on screen. I can only tap to click.

## What I've already tried
Switch USB ports. 
Restarting OTD. 
Restarting my PC.
Switching stylus nubs.

## Diagnostic Information

Mrcubix, 2021-06-21:
Do you see that one program on your desktop called "Wacom", head to programs and feature and uninstall that _**thing**_ from your PC.

Lorphex, 2021-06-21:
I don't have the default Wacom drivers / program installed

Mrcubix, 2021-06-21:
Then fill the template and provide a diagnostic file : Help > export diagnostic...

Lorphex, 2021-06-21:
The only files when I do that are .JSON and that type isn't supported by github attachments

Mrcubix, 2021-06-21:
just zip it or .txt it?

Lorphex, 2021-06-21:
Ran the updater and it seems to work now. Updater shows I'm on 0.5.3.2, but in OTD itself it shows I'm in 0.5.3.1 still.
Had the same issue as #1198

Mrcubix, 2021-06-21:
that person never filled the template so we will never know whether or not they have the same issue, a good amount of new users have messed up with their installs in different way which require different method to fix issues.

since your issue seem solved, i'll close.

closed 2021-06-21 by Mrcubix