OpenTabletDriver/OpenTabletDriver #1208: Tablet causes cursor to spaz

ryynison, 2021-06-21:
## Description

Tablet causes cursor to spaz out both while dragging and hovering as seen below. I have followed the instructions to blacklist the kernel wacom drivers already.

https://user-images.githubusercontent.com/25913782/122712045-206d1580-d218-11eb-82b3-996af6e7c60b.mp4

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Arch Linux x86_64
| Software Version | v0.5.3.2 (Latest release)
| Tablet           | Wacom CTH-680

ryynison, 2021-06-21:
It was the wireless charging pad built into my desk all along 🤦

closed 2021-06-21 by ryynison

InfinityGhost, 2021-06-21:
Yeah that makes sense, the wireless charging pad uses EMR just like the pen so it will send garbage data with something like that.