OpenTabletDriver/OpenTabletDriver #1209: Support Request:

Henovic, 2021-06-21:
# Support Request
 When i connect my Wacom CTL-4100 the driver detects the tablet but when my pen makes contact with the surface of the tablet it doesnt detect it anymore

## What I've already tried
 uninstalling and reinstalling, disconnecting and reconnecting the tablet, plugging into different USB port, restarting PC

## Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.5.3.1",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19042.0",
    "VersionString": "Microsoft Windows NT 10.0.19042.0"
  },
  "Environment Variables": {
    "ComSpec": "C:\\WINDOWS\\system32\\cmd.exe",
    "USERDOMAIN_ROAMINGPROFILE": "DESKTOP-94GL4U2",
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules",
    "ProgramData": "C:\\ProgramData",
    "COMPUTERNAME": "DESKTOP-94GL4U2",
    "PROCESSOR_IDENTIFIER": "AMD64 Family 23 Model 17 Stepping 0, AuthenticAMD",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "USERPROFILE": "C:\\Users\\moody",
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "APPDATA": "C:\\Users\\moody\\AppData\\Roaming",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "LOGONSERVER": "\\\\DESKTOP-94GL4U2",
    "TEMP": "C:\\Users\\moody\\AppData\\Local\\Temp",
    "PROCESSOR_LEVEL": "23",
    "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC",
    "OneDriveConsumer": "C:\\Users\\moody\\OneDrive",
    "OneDrive": "C:\\Users\\moody\\OneDrive",
    "NUMBER_OF_PROCESSORS": "4",
    "USERDOMAIN": "DESKTOP-94GL4U2",
    "USERNAME": "moody",
    "SystemRoot": "C:\\WINDOWS",
    "SystemDrive": "C:",
    "Path": "C:\\Program Files (x86)\\Common Files\\Oracle\\Java\\javapath;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Program Files\\dotnet\\;C:\\Users\\moody\\AppData\\Local\\Microsoft\\WindowsApps;;C:\\Users\\moody\\AppData\\Local\\Programs\\Microsoft VS Code\\bin",
    "ProgramFiles": "C:\\Program Files",
    "ProgramW6432": "C:\\Program Files",
    "LOCALAPPDATA": "C:\\Users\\moody\\AppData\\Local",
    "TMP": "C:\\Users\\moody\\AppData\\Local\\Temp",
    "windir": "C:\\WINDOWS",
    "HOMEDRIVE": "C:",
    "PUBLIC": "C:\\Users\\Public",
    "SESSIONNAME": "Console",
    "HOMEPATH": "\\Users\\moody",
    "OPENSSL_ia32cap": "0x20000000",
    "PROCESSOR_REVISION": "1100",
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "OS": "Windows_NT"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_0c45&pid_7631&mi_01&col04#8&1d60aa10&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 3141,
      "ProductID": 30257,
      "CanOpen": true,
      "ReleaseNumberBcd": 262,
      "ReleaseNumber": "1.6",
      "ProductVersion": 262,
      "Manufacturer": "SONiX",
      "ProductName": "USB DEVICE",
      "SerialNumber": "",
      "MaxInputReportLength": 64,
[... 1296 lines cut ...]
      "Time": "2021-06-21T15:44:45.7557692+01:00",
      "Group": "Detect",
      "Message": "Using report parser type 'OpenTabletDriver.Vendors.Wacom.IntuosV3ReportParser'.",
      "Level": 0
    },
    {
      "Time": "2021-06-21T15:44:45.7557785+01:00",
      "Group": "Detect",
      "Message": "Device path: \\\\?\\hid#vid_056a&pid_0374#7&32485f2f&1&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Level": 0
    },
    {
      "Time": "2021-06-21T15:44:45.7557879+01:00",
      "Group": "Detect",
      "Message": "Found tablet 'Wacom CTL-4100'",
      "Level": 1
    },
    {
      "Time": "2021-06-21T15:44:45.7557983+01:00",
      "Group": "NullReferenceException",
      "Message": "Object reference not set to an instance of an object.",
      "Level": 3
    }
  ]
}

Mrcubix, 2021-06-21:
Fill the template by providing a diagnostic file

Henovic, 2021-06-21:
> Fill the template by providing a diagnostic file

its a JSON file i cant it says not supported

Mrcubix, 2021-06-21:
then zip it or .txt it???

Henovic, 2021-06-21:
> then zip it or .txt it???

[fix.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6688073/fix.zip)

Mrcubix, 2021-06-21:
File > reset to default, apply if this doesn't fix it, re-download OTD (portable) and try again

Henovic, 2021-06-21:
> File > reset to default, apply. If this doesn't fix it, re-download OTD (portable) and try again

its fixed now thank you

closed 2021-06-21 by Henovic