OpenTabletDriver/OpenTabletDriver #1213: No tablet found.

Niron68, 2021-06-22:
## Description

No tablet found. Same if i try the "Detect tablet" option

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Pop os 20.10 |
| Software Version | 0.5.3.1 |
| Tablet           | CTL-4100 |

jjhhgg100123, 2021-06-22:
First check the console and compare it to the [Linux wiki](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ).

If no errors in console match up, export diagnostics. And plop it here. I can almost guarantee it's a combo of 1/2 in the wiki.

Niron68, 2021-06-22:
> First check the console and compare it to the [Linux wiki](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ).
> 
> If no errors in console match up, export diagnostics. And plop it here. I can almost guarantee it's a combo of 1/2 in the wiki.

How do i export diagnostics ?

Niron68, 2021-06-22:
> First check the console and compare it to the [Linux wiki](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ).
> 
> If no errors in console match up, export diagnostics. And plop it here. I can almost guarantee it's a combo of 1/2 in the wiki.

How do i send you the diagnostic file ?

jjhhgg100123, 2021-06-22:
Try dropping this config in your .config OTD Config folder, you may have a weird variant.
[CTL-4100.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6692358/CTL-4100.zip)

Niron68, 2021-06-22:
> Try dropping this config in /usr/share/OpenTabletDriver folder, you may have a weird variant.
> [CTL-4100.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6692358/CTL-4100.zip)

Still got "No tablet found" with this config file. The exact model of my tablet is CTL-4100K-S, is it the tablet ?

SnowLire, 2021-06-22:
Your tablet is in Android mode at the moment, do [this](https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#android-misdetect) troubleshooting step from libwacom and place this old CTL-4100 config back into /usr/share/OpenTabletDriver/Configurations/Wacom
[CTL-4100.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6692514/CTL-4100.zip)

Niron68, 2021-06-22:
> Your tablet is in Android mode at the moment, do [this](https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#android-misdetect) troubleshooting step from libwacom and place this old [...]

Now, i'm back with only the "No tablet found".

SnowLire, 2021-06-22:
Can you join the [Discord server](discord.gg/p3gydus)? It'll be easier to communicate there compared to Github comments.

Niron68, 2021-06-22:
> Can you join the [Discord server](discord.gg/p3gydus)? It'll be easier to communicate there compared to Github comments.

The invite is invalid or expired

SnowLire, 2021-06-22:
> The invite is invalid or expired

https://discord.gg/P3GYdUS
_stupid capitalisation_

Kuuuube, 2021-06-22:
Closing this as it appears to have been resolved in discord.

closed 2021-06-22 by Kuuuube