OpenTabletDriver/OpenTabletDriver #1214: Tablet driver breaks when i open it before plugging in my tablet

Thaanuu, 2021-06-22:
## Description

When I open the drivers before plugging in my tablet and then when i plugged in my tablet and press with the pen on it this reload message shows up:
[image]

The console says " ([Plugin:Error]    Unable to construct object 'TabletDriverFilters.Hawku.Smoothing') " 
when i start the driver before i plugged in my tablet. 

But everything works normally if i plug in the tablet before i start the driver but its annoying to do that everytime cause i cant run it at startup and have the drivers in the background.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System |0.5.3.1 (Latest Release)
| Software Version |0.5.3.2 (Latest Release)
| Tablet           | Gaomon S620

InfinityGhost, 2021-06-22:
1. <kbd>Win</kbd>+<kbd>R</kbd>
2. Type in `eventvwr` and click `Ok`
3. On the left side of the event viewer window, navigate to `Windows Logs` -> `Application`
[image]
4. On the right side of the event viewer window, click on `Filter Current Log...`
[image]
5. Change `Event sources` to `.NET Runtime` in the window that pops up, then click `Ok`
[image]
6. Select the first error in the list of events in the main event viewer window, and copy the text under the general tab
[image]

Thaanuu, 2021-06-22:
Application: OpenTabletDriver.Daemon.exe
CoreCLR Version: 5.0.721.25508
.NET Version: 5.0.7
Description: The process was terminated due to an unhandled exception.
Exception Info: System.NullReferenceException: Object reference not set to an instance of an object.
   at TabletDriverFilters.Hawku.Smoothing.Interpolate()
   at OpenTabletDriver.Plugin.Tablet.Interpolator.Interpolator.InterpolateHook()
   at OpenTabletDriver.Desktop.Interop.Timer.WindowsTimer.Callback(UInt32 uTimerID, UInt32 uMsg, UIntPtr dwUser, UIntPtr dw1, UIntPtr dw2)

InfinityGhost, 2021-06-22:
Delete the folder at `%LOCALAPPDATA%\OpenTabletDriver\Plugins`

Thaanuu, 2021-06-22:
I did it, but after installing the plugins(Hawku smoothing and antichatter exterminator) again, i get the same reload bug again

InfinityGhost, 2021-06-23:
Odds are that something is broken in the plugin. I'm going to transfer this issue to the correct repository.