OpenTabletDriver/OpenTabletDriver #1215: .NET v5 runtime installed but not detected

numericOverflow, 2021-06-23:
## Description
I've installed .NETv5 (both x86 & x64 versions) yet it's not detected when I try to startup the driver.  It displays the error dialog asking me to download now.

How does the driver detect .NETv5 is installed? I've confirmed v5 is installed, and everything I've done suggests it's installed (see screenshots below).  I've tried bot the installer and portable versions of OpenTabletDriver, both function the same and give me the error about not having .NETv5 installed.

Thoughts?

I can provide additional debug info if needed/helpful, just let me know what you'd like to see.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Kernel version:            Windows 10 Enterprise
||Product type:              Professional
||Product version:           6.3
||Service pack:              0
||Kernel build number:       18363
| Software Version | 0.5.3.2
| Tablet           | Veikk A30

[image]
[image]
[image]
[image]

Mrcubix, 2021-06-23:
AFAIK, OTD isn't only a console application so you need the right .NET version which is obviously ***DEKSTOP RUNTIME*** https://dotnet.microsoft.com/download/dotnet/thank-you/runtime-desktop-5.0.7-windows-x64-installer

closed 2021-06-23 by Mrcubix

numericOverflow, 2021-06-23:
@Mrcubix - That was exactly right.  I downloaded & installed from the link you included above and it is working perfectly so far.

I didn't realize there were console and desktop versions of .NETv5.

Thanks!

Mrcubix, 2021-06-23:
Microsoft prompt is very bad in the sense that it doesn't tell one which one to download from one of these 3 if you land on this page :
[image]
(https://dotnet.microsoft.com/download/dotnet/5.0/runtime)
or this page :
[image]
(what you need there is in red on the right)
pretty confusing for new users ngl