OpenTabletDriver/OpenTabletDriver #1216: Support Request:

w-i-s-n, 2021-06-24:
# Support Request
Windows Version: windows 10

Tablet Driver version: 0.5.3.2

Tablet model: veikk s640

everything worked perfectly before, during that time i was using the official veikk drivers, but then i feel like the settings are lacking so i uninstalled the veikk drivers and installed the hawku drivers as it was the first one on a google search. it won't work as it tells me that the vmulti device wont work or something, so i went back to veikk drivers and it kinda messed up something that now whenever an administrator enabled window is on top the tablet won't move the mouse. Then i searched for helped on hawku issues and saw that it was already abandoned so i moved here and maybe opentabletdriver will help me fix this issue

## What I've already tried
i've tried reinstalling. also tried uninstalling hawku multiple times and looked for it on the directory stated from the cmd, also checked driverquery as it's no longer there.

## Diagnostic Information

jjhhgg100123, 2021-06-24:
We don't support Hawku, use OpenTabletDriver.

w-i-s-n, 2021-06-24:
nonooo. im currently using opentabletdriver. it's just that everything was running good untill i installed hawku. i uninstalled it already but the problem still persists. is there anyway to get it to work on admistrative window?

SnowLire, 2021-06-24:
OpenTabletDriver is a user-mode driver so by default it cannot send input if an admin app/window is in focus. You can try using VMultiMode or Windows Ink Plugin but VMulti Absolute Mode suffers from the same issue as Hawku's absolute mode. Read the plugin wiki for how to set it up.

w-i-s-n, 2021-06-24:
how come the veikk driver worked? i believe installing hawku kind of made all tablet inputs use user-mode driver, even if i used the veikk drivers back it wont work still

SnowLire, 2021-06-24:
Probably conflicting VMulti version, since both Veikk and Hawku uses it. VMulti itself is a kernel driver so it can send input regardless of which app is in focus.

w-i-s-n, 2021-06-24:
is there any way to reinstall it? i see the vmulti from driverquery

SnowLire, 2021-06-24:
Well, first uninstall any other vmulti driver you have on your PC, then follow the steps on the VMultiMode plugin wiki to get the correct vmulti installed and set up the plugin itself.

w-i-s-n, 2021-06-24:
ill try it out when i get home thanks

X9VoiD, 2021-06-24:
Veikk worked because Veikk uses VMulti

w-i-s-n, 2021-06-25:
hello i am out of ideas. can anyone help me uninstall the vmulti.inf driver package, as google searches have failed me and that driver is not visible in pnputil, but i can see it when running driver query

InfinityGhost, 2021-10-30:
Stale support request, assuming it has been resolved. This can be reopened in the case this was a mistake.

closed 2021-10-30 by InfinityGhost