OpenTabletDriver/OpenTabletDriver #1217: Artist mode does not work properly in wayland (KDE)

aika-64, 2021-06-24:
## Description

The pen pressure doesn't work in absolute mode, so I switched to artist mode, but I can't even click with the pen
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Arch Linux
| Software Version | v0.5.3.2
| Tablet           | Gaomon M10K

InfinityGhost, 2021-06-24:
Are you using the `opentabletdriver-git` package or the versioned `opentabletdriver` package?

aika-64, 2021-06-24:
the -git package

InfinityGhost, 2021-06-26:
Use the `opentabletdriver` package rather than the git version, as how it works is actively being changed.

gonX, 2021-07-08:
Have you set your pen (and eraser, if applicable) tip binding to the Pen Passthrough binding? If you don't, nothing will happen.

[image]

InfinityGhost, 2021-07-18:
Resolved in latest master.

closed 2021-07-18 by InfinityGhost

ChristianHein, 2021-09-14:
> If you're still on master, have you set your pen (and eraser, if applicable) tip binding to the Pen Passthrough binding? If you don't, nothing will happen.

Please add this information to the project docs on your next release, so others can find this.

The main reason I use Artist Mode is for pen pressure support. Because this works in v0.5.3.3 without additional configuration, I had assumed that the lastest master was bugged because my pen stopped applying pressure entirely in Artist Mode. Spent quite a bit of time trying to figure out what was wrong.

gonX, 2021-11-13:
> this works in v0.5.3.3 without additional configuration

If I remember correctly, 0.5.x did not have correctly working bindings when in artist mode.

There will likely be guides on how to set it up by the time the 0.6 release rolls around, but it sounds like we need to track this in a separate issue similar to #1585 except for Linux. An idea could be to notify the user if we're in artist mode and Pen Passthrough is not bound.

aika-64, 2021-11-14:
> If you're still on master, have you set your pen (and eraser, if applicable) tip binding to the Pen Passthrough binding? If you don't, nothing will happen.

[image]