OpenTabletDriver/OpenTabletDriver #1219: Support Request:

w-i-s-n, 2021-06-25:
# Support Request
i could not run move the pen when using vmulti mode

## What I've already tried
i have already installed vmulti driver from the link provided as stated on the console yet still the same error 

## Diagnostic Information
[VMultiRel:Error]	Cannot find VirtualHID
[VMultiRel:Error]	Install VMulti driver here: https://github.com/X9VoiD/vmulti-bin/releases/latest
[Plugin:Error]	Unable to construct object 'VoiDPlugins.OutputMode.VMultiRelativeMode'
[Settings:Info]	Tip Binding: [Mouse Button Binding: Left]@1%
[Settings:Info]	Pen Bindings: [0, Mouse Button Binding: Right], [1, Mouse Button Binding: Middle]
[Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ], [6, ], [7, ]
[Settings:Info]	Driver is auto-enabled.

X9VoiD, 2021-06-25:
Uninstall all other VMultis that you have installed. Will close as this is not an issue within OTD itself

closed 2021-06-25 by X9VoiD

w-i-s-n, 2021-06-25:
how to? i think you've seen me already in multiple threads. please teach me how because i already uninstalled vmulti and deleted it also from drivers directory, also tried to remove it using the link above still no hope