OpenTabletDriver/OpenTabletDriver #1220: Add support for XP-PEN Deco Fun S

TheVoidKitsune0, 2021-06-26:
# Support Request
Please add support for XP-pen deco fun S, 
I've looked through the supported tablet list and there is no record of it in there,
 even tried downloading and running OTD with no luck.

## Diagnostic Information
[Dia.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6719801/Dia.zip)

InfinityGhost, 2021-07-01:
## Specifications

[image]

InfinityGhost, 2021-07-02:
<details>
  <summary>
      Potential Devices
  </summary>
  <div class="snippet-clipboard-content position-relative">
    <pre>
{
  "DevicePath": "\\\\?\\hid#vid_28bd&pid_0931&mi_00&col01#7&2ae44e78&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
  "VendorID": 10429,
  "ProductID": 2353,
  "CanOpen": false,
  "ReleaseNumberBcd": 0,
  "ReleaseNumber": "0.0",
  "ProductVersion": 0,
  "Manufacturer": "UGTABLET",
  "ProductName": "6 inch PenTablet",
  "SerialNumber": "000000",
  "MaxInputReportLength": 8,
  "MaxOutputReportLength": 0,
  "MaxFeatureReportLength": 0
},
{
  "DevicePath": "\\\\?\\hid#vid_28bd&pid_0931&mi_00&col02#7&2ae44e78&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
  "VendorID": 10429,
  "ProductID": 2353,
  "CanOpen": false,
  "ReleaseNumberBcd": 0,
  "ReleaseNumber": "0.0",
  "ProductVersion": 0,
  "Manufacturer": "UGTABLET",
  "ProductName": "6 inch PenTablet",
  "SerialNumber": "000000",
  "MaxInputReportLength": 8,
  "MaxOutputReportLength": 0,
  "MaxFeatureReportLength": 0
},
{
  "DevicePath": "\\\\?\\hid#vid_28bd&pid_0931&mi_00&col03#7&2ae44e78&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
  "VendorID": 10429,
  "ProductID": 2353,
  "CanOpen": false,
  "ReleaseNumberBcd": 0,
  "ReleaseNumber": "0.0",
  "ProductVersion": 0,
  "Manufacturer": "UGTABLET",
  "ProductName": "6 inch PenTablet",
  "SerialNumber": "000000",
  "MaxInputReportLength": 8,
  "MaxOutputReportLength": 0,
  "MaxFeatureReportLength": 0
},
{
  "DevicePath": "\\\\?\\hid#vid_28bd&pid_0931&mi_01#7&279a0af&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
  "VendorID": 10429,
  "ProductID": 2353,
  "CanOpen": false,
  "ReleaseNumberBcd": 0,
  "ReleaseNumber": "0.0",
  "ProductVersion": 0,
  "Manufacturer": "UGTABLET",
  "ProductName": "6 inch PenTablet",
  "SerialNumber": "000000",
  "MaxInputReportLength": 10,
  "MaxOutputReportLength": 0,
  "MaxFeatureReportLength": 0
},
{
  "DevicePath": "\\\\?\\hid#vid_28bd&pid_0931&mi_02#7&15a9bced&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
  "VendorID": 10429,
  "ProductID": 2353,
  "CanOpen": true,
  "ReleaseNumberBcd": 0,
  "ReleaseNumber": "0.0",
  "ProductVersion": 0,
  "Manufacturer": "UGTABLET",
  "ProductName": "6 inch PenTablet",
  "SerialNumber": "000000",
  "MaxInputReportLength": 12,
  "MaxOutputReportLength": 12,
  "MaxFeatureReportLength": 0
}</pre>
  </div>
</details>

InfinityGhost, 2021-07-17:
Finally got around to adding support for this tablet. Please download the binaries [here](https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/1041279514) and let me know how it goes.

TheVoidKitsune0, 2021-07-18:
> 
> 
> Finally got around to adding support for this tablet. Please download the binaries [here](https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/1041279514) and let me know how it goes.
I seem to have found a bug with the display settings I don't know if it's because of the tablet or the application.

https://user-images.githubusercontent.com/85374901/126079402-4ca22335-11a0-4f5f-8d81-8cfe2eb5ac1f.mp4

it happens If I set it to a single display instead of both.

TheVoidKitsune0, 2021-07-18:
> 
> 
> > Finally got around to adding support for this tablet. Please download the binaries [here](https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/1041279514) and let me know how it goes.
> 
> I seem to have found a bug with the display settings I don't know if it's because of the tablet or the application.
> OpenTabletDriver.v0.5.3.1.-.XP-Pen.CT460.2021-07-18.15-04-29.mp4
> 
> it happens If I set it to a single display instead of both.
> 
> > UPDATE
> 
> Even happens if I change the tablet area

Seem to have fixed the issue, I had to uninstall my mouse then reinstall it overall, everything seems to be working fine.

TheVoidKitsune0, 2021-07-19:
I’m having an issue with plugins, I can install them but they don’t show up.

InfinityGhost, 2021-07-19:
The build I have linked does not work with the existing plugins in the plugin repository, which would explain your issue. However all i needed was verification that it is functional, which you reported it is so we can merge it in for v0.6.0

PR #1272 (2021-07-19): Add support for the XP-Pen CT460
## Verification
- [x] XP-Pen CT460 (in #1220)

## Issues
- Closes #1220
closed 2021-07-19 by InfinityGhost