OpenTabletDriver/OpenTabletDriver #1222: Support Request: Gaomon S620 not being detected.

Aiptex, 2021-06-26:
# Support Request
My Gaomon S620 tablet is not getting detected by the software.

## What I've already tried
https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Windows-FAQ#manual-cleanup
https://github.com/OpenTabletDriver/OpenTabletDriver/issues/405

## Diagnostic Information
[a.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6720544/a.txt)

Mrcubix, 2021-06-26:
you said you tried https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Windows-FAQ#manual-cleanup except i still see vmulti and WinUSB installed.

https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0

run the script, should work.

Aiptex, 2021-06-26:
I have ran the script and reconnected my tablet, but my tablet is still not being detected.

[b.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6720640/b.txt)

Mrcubix, 2021-06-26:
run it as admin again, if it still doesn't work, right click on start button > device manager > view > device by container > find your tablet in there, open the container, screenshot device manager and upload it here

Aiptex, 2021-06-26:
I tried running it as admin, but a command window appears then instantly disappears.

[image]

Mrcubix, 2021-06-26:
uninstall razer synapse

Mrcubix, 2021-06-26:
after that, right click that virtual HID minidriver collection > update > let me choose from a list > usb input device

Aiptex, 2021-06-26:
I don't have razer synapse installed

[image]

Mrcubix, 2021-06-26:
HID compliant device

Aiptex, 2021-06-26:
It works now, thank you very much.

closed 2021-06-26 by Aiptex