OpenTabletDriver/OpenTabletDriver #1223: Support Request: How to properly map tablet? Coming from TabletDriver to here.

TheRedEyedSpecter, 2021-06-29:
# Support Request
 I more or less would like some help with mapping my tablet? The tablet is supported, I already checked, but coming from using TabletDriverGUI I am used to that kind of mapping instead. I have 2 monitors, each with a different resolution. I would like to use the monitor with the higher resolution only, being 1920x1080. When I used the mapping of TabletDriver, I had the screen mapping set to 1920x1080 with left offset of 1360, and top offset of 0, just because that is how the mapping had to be for me to use my entire one monitor screen. For thetablet area, I had it set to 88.88x50 with x=76.91 and y=40.84 since that is where I wanted the mapping of my tablet area to be, since I did not want to use the full tablet area. In TabletDriver, the mapping was exactly where it needed to be with those numbers, but since I am not used to OpenTabletDriver, the mapping is a little off since my mouse extends past my monitor screen whereas it would hit the border and go no further with TabletDriver. Is there a proper way to map this or am I missing something here?

[image]

## What I've already tried
 I tried browsing the FAQ sections but honestly I was pretty confused on how to navigate it since I'm new to GitHub.

## Diagnostic Information

If something is needed here I will attach it later somehow since it says "We don't support that file type."

Mrcubix, 2021-06-29:
first of all, if you're gonna use the tablet area offset you had in TD, TD has their offset starting from the top left corner of the area whereas OTD has there offset in the middle of the area, so to achieve, the same offset than in TD, you will need to do the following if no already done:

offset X = offset X in TD + (area width/ 2)
offset Y = offset Y in TD + (area height / 2)

as for your second issue being your cursor being able to go past your display area, just to make sure that's out of the way, uninstall wacom drivers via "add of remove a program" if not already done, then make sure you actually saved your setting, and finally make sure area clipping is enabled. let us know if the issue still persist.

X9VoiD, 2021-06-29:
Right click -> Set to display

closed 2021-06-29 by X9VoiD