OpenTabletDriver/OpenTabletDriver #1226: Plugin Manager not opening

WarmestDoughnut, 2021-06-29:
## Description

So the first interesting thing to note is while installing this with a friend his worked fine but mine whenever I open the portable version(latest) it gives me an error message saying the file should not be run with administrator access as some features will be unavailable and I should enable UAC to resolve this(I am not running the file in administrator mode and I have tried enabling UAC and launching the driver did not change the error message). This in itself is not the key issue but might be the cause. The issue is the plugin manager does not open at all. No matter how many times I spam the button or just a single click there is no change and nothing opens up. I have checked task manager when I spam the plugin manager button and the CPU usage goes up for the program but nonetheless goes down after I stop. I've tried reinstalling several times same issue. Can't get the plugin manager to open.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10 pro 64-bit
| Software Version | Opentablet portable latest version
| Tablet           | Wacom CTL-4100

X9VoiD, 2021-06-29:
If you are not running it as Administrator, have enabled UAC, have restarted the PC, then it should be already fine. But if the warning at the beginning is still present then make sure that you are not running Window's hidden **Administrator** account. That account is not suitable for daily use, breaks software that relies on **not** being run with elevated permissions, and is literally the greatest security risk as it allows *all programs to have elevated permissions*.

Create an actual user account with administrator capabilities as a fix if the above situation applies. Will close issue as this is not an issue within OTD itself.

closed 2021-06-29 by X9VoiD

SnowLire, 2021-06-29:
You are most likely running Windows as the Administrator account, which runs every program implicitly as Admin (similar to how turning UAC to off works in Windows 10). Follow these steps:

- Create an account with no perms (if you don't have one)
- Move OTD to that account
- On the non-admin account, create a new .bat file, insert inside the following:
```cmd
cd [path to OTD folder]
start OpenTabletDriver.UX.Wpf.exe
```
- Create a shortcut, in the path, paste the following: 
```cmd
"C:\Windows\System32\cmd.exe" /k runas /user:[enter non-admin account name here] [path to previous batch file]
```
- Go into the property of the existing osu! shortcut (if you have one) and change the location to 
```cmd
"C:\Windows\System32\cmd.exe" /k runas /user:[enter non-admin account name here] [path to osu.exe here]
```

Remember to replace all the bracketed arguments with their corresponding value. You may also need to use "quotes" for file paths.