OpenTabletDriver/OpenTabletDriver #1230: Add artist mode for Windows

UlyssesZh, 2021-07-01:
The official Wacom driver for Windows offers an option not to use Windows Ink but it can still provide pressure. I think there should also be an artist mode in OTD for Windows that does not have anything to do with Windows Ink. Using Windows Ink will lead to too many side effects.

AbstractQbit, 2021-07-01:
Wintab is is an ancient API and an absolute nightmare to implement. Really doubt it will ever be supported.

adryzz, 2021-07-01:
> Wintab is is an ancient API and an absolute nightmare to implement. Really doubt it will ever be supported.

And it would also 99% require you to add a DLL library in your drawing software or in system32 (unless we use Wacom's, which is of course undocumented too)

fredemmott, 2022-03-04:
Is there an alternative API/output mode that applications could be modified to use to get data from OpenTabletDriver, without going via windows ink or mouse emulation?

In my case, I'm looking for an alternative to huion's wintab drivers that I could add to https://github.com/fredemmott/openkneeboard ; for this use case, it would be an added bonus if background access were supported - currently it uses DLL injection and SetWindowLongPtrA for wintab when another application is active.

From a brief look, it seems that this is possible to add, but would need a new plugin to be written first to add this functionality?

jamesbt365, 2022-03-04:
This issue is on about adding wintab support.

fredemmott, 2022-03-04:
Sorry, split to separate issue.

GlitchyPSIX, 2024-05-31:
Is this still being considered? It has been expressed before that Wintab is hard to implement, but there doesn't seem to be further discussion on why is it so. I am unfamiliar with the API, so I'd just like to know any potential hurdles.

fredemmott, 2024-05-31:
https://github.com/OpenKneeboard/OTD-IPC/ is a plugin with a working alternative to Wintab for detailed exclusive access, but it would require applications to add support.

GlitchyPSIX, 2024-05-31:
I see.
Mmm. This would make it tricky for many commercial applications.

jamesbt365, 2024-05-31:
WinTab is ancient at this point and legacy, for what its worth I don't think we'll ever really support it, we are better off putting resources towards making Windows Ink easier to set up, possibly even installed by default, or providing a button to set it up.

Windows Ink is newer, typically better supported but isn't perfect, but WinTab isn't either.

WinTab is a lot more sketch to use and OTD will likely trip your antivirus hard if you were to try and use WinTab with it as we must inject a DLL into the drawing program, our application is not signed and this just puts hurdle after hurdle here.

If you must use an application that does not have any Windows Ink support, or the support is subpar compared to the WinTab implementation, you are better off not using OTD for this. WinTab just doesn't seem feasible to implement at this time, maybe at all.

GlitchyPSIX, 2024-05-31:
I see, thanks for the input!

I'm working in an app to aid with VTuber tablet visualizations and this needs to track pen position, WinInk does not update the mouse position (in my experience) so common ways of tracking the mouse are off-limits. 
May make an OTD plugin to aid for this purpose, but I feel that I'd be hijacking this issue, so I'll stop here about it 😛