OpenTabletDriver/OpenTabletDriver #1235: Support Request: Help, something is wrong with my tablet and my pc's connection

Inner574, 2021-07-03:
# Support Request

First, OTD doesn't detects my Gaomon 1060 Pro(No tablet detected). Works very fine with their(Gaomon) drivers.
If I plug it in, the windows device plug-in sound plays, but the disconnect sound plays right after the sound. still works with Gaomon's driver.
Tablet stops working when I press a button on it
## What I've already tried

Uninstalling OpenTabletDriver and reinstalling
Installing their drivers and running OTD and the driver at the same time
Uninstalling all my tablet drivers and restarting
Looking at the Bluetooth & other devices, Device Manager (Windows settings, If I plug in the tablet, 'USB Composite Device' appears, idk if its normal.)
Searching if there is a same problem in the issues
Trying the 'Detect' option multiple times
## Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.5.3.1",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19041.0",
    "VersionString": "Microsoft Windows NT 10.0.19041.0"
  },
  "Environment Variables": {
    "SystemDrive": "C:",
    "ComSpec": "C:\\WINDOWS\\system32\\cmd.exe",
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules",
    "NUMBER_OF_PROCESSORS": "8",
    "APPDATA": "C:\\Users\\hwang\\AppData\\Roaming",
    "LOCALAPPDATA": "C:\\Users\\hwang\\AppData\\Local",
    "COMPUTERNAME": "DESKTOP-33TEO81",
    "GTK_BASEPATH": "C:\\Program Files (x86)\\GtkSharp\\2.12\\",
    "PROCESSOR_LEVEL": "6",
    "USERDOMAIN": "DESKTOP-33TEO81",
    "PUBLIC": "C:\\Users\\Public",
    "ChocolateyInstall": "C:\\ProgramData\\chocolatey",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "FPS_BROWSER_USER_PROFILE_STRING": "Default",
    "USERPROFILE": "C:\\Users\\hwang",
    "SESSIONNAME": "Console",
    "USERDOMAIN_ROAMINGPROFILE": "DESKTOP-33TEO81",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "SystemRoot": "C:\\WINDOWS",
    "COMPOSE_CONVERT_WINDOWS_PATHS": "true",
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "GOPATH": "C:\\Users\\hwang\\go",
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "OS": "Windows_NT",
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "USERNAME": "hwang",
    "HOMEPATH": "\\Users\\hwang",
    "LOGONSERVER": "\\\\DESKTOP-33TEO81",
    "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC;.PY;.PYW",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "ProgramData": "C:\\ProgramData",
    "PROCESSOR_IDENTIFIER": "Intel64 Family 6 Model 60 Stepping 3, GenuineIntel",
    "TEMP": "C:\\Users\\hwang\\AppData\\Local\\Temp",
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "TMP": "C:\\Users\\hwang\\AppData\\Local\\Temp",
    "CapMonsterCurrentPath": "C:\\Users\\hwang\\Desktop\\capmonster",
    "JD2_HOME": "C:\\Users\\hwang\\AppData\\Local\\JDownloader 2.0",
    "VS140COMNTOOLS": "C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\Common7\\Tools\\",
    "FPS_BROWSER_APP_PROFILE_STRING": "Internet Explorer",
    "OneDrive": "C:\\Users\\hwang\\OneDrive",
    "HOMEDRIVE": "C:",
    "ProgramFiles": "C:\\Program Files",
    "ChocolateyLastPathUpdate": "132402874376534978",
    "Path": "C:\\Users\\hwang\\AppData\\Local\\Programs\\Python\\Python39\\Scripts\\;C:\\Users\\hwang\\AppData\\Local\\Programs\\Python\\Python39\\;C:\\Program Files (x86)\\Common Files\\Oracle\\Java\\javapath;C:\\Program Files\\AdoptOpenJDK\\jdk-8.0.275.1-hotspot\\bin;C:\\Program Files (x86)\\VMware\\VMware Workstation\\bin\\;C:\\Python38\\Scripts\\;C:\\Python38\\;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\NVIDIA Corporation\\NVIDIA NvDLISR;C:\\Program Files\\dotnet\\;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\ProgramData\\chocolatey\\bin;C:\\Program Files\\nodejs\\;C:\\Program Files\\Git\\cmd;C:\\Program Files\\Microsoft SQL Server\\130\\Tools\\Binn\\;C:\\Program Files\\Go\\bin;C:\\Program Files (x86)\\GtkSharp\\2.12\\bin;C:\\Users\\hwang\\AppData\\Local\\Programs\\Python\\Python38\\Scripts\\;C:\\Users\\hwang\\AppData\\Local\\Programs\\Python\\Python38\\;C:\\Users\\hwang\\AppData\\Local\\Microsoft\\WindowsApps;;C:\\Users\\hwang\\AppData\\Roaming\\npm;C:\\Users\\hwang\\.dotnet\\tools;C:\\Users\\hwang\\go\\bin;C:\\Users\\hwang\\AppData\\Local\\Programs\\Microsoft VS Code\\bin",
    "ProgramW6432": "C:\\Program Files",
    "VBOX_MSI_INSTALL_PATH": "C:\\Program Files\\Oracle\\VirtualBox\\",
[... 6812 lines cut ...]
      "Level": 1
    },
    {
      "Time": "2021-07-03T17:47:14.6868567+09:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'Huion GC610'",
      "Level": 1
    },
    {
      "Time": "2021-07-03T17:47:14.6868614+09:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'Huion H1060P'",
      "Level": 1
    },
    {
      "Time": "2021-07-03T17:47:14.686866+09:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'Huion H1161'",
      "Level": 1
    },
    {
      "Time": "2021-07-03T17:47:14.6868708+09:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'Huion H430P'",
      "Level": 1

SnowLire, 2021-07-07:
Plug you tablet directly into your PC and not through a USB hub.

Next follow the steps [here](https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0) to remove the residue from the Gaomon driver.

Inner574, 2021-07-08:
> Plug you tablet directly into your PC and not through a USB hub.
> 
> Next follow the steps [here](https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0) to remove the residue from the Gaomon driver.

I did, still not detecting it. Windows detects it well and calls it 'PenTablet', so I ran the WinUSBCleanup again, still not working. Should I attach another diagnostic file?

SnowLire, 2021-07-08:
> I did, still not detecting it. Windows detects it well and calls it 'PenTablet', so I ran the WinUSBCleanup again, still not working. Should I attach another diagnostic file?

That would help yes

Inner574, 2021-07-09:
[diag.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6787915/diag.zip)

Inner574, 2021-07-10:
Closing this issue because I found out my 1060pro isn't supported.

closed 2021-07-10 by Inner574