OpenTabletDriver/OpenTabletDriver #1238: OTD crashes/doesn't detect XP Pen G640

BramboraSK, 2021-07-04:
## Description
I've downloaded OTD using AUR, I've tried downloading opentabletdriver-git, but it didn't detect my tablet no matter what, then I uninstalled it and installed opentabletdriver, but it crashes at Connecting to OpenTabletDriver daemon. I've tried to export diagnostics, but it exported a blank file.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | EndeavourOS Linux x86_64
| Software Version | v0.5.3.1-r315.g1cb3d12a
| Tablet           | XP-Pen Star G640

BramboraSK, 2021-07-04:
I can see my tablet in `lsusb`:
`Bus 001 Device 009: ID 28bd:0094 XP-Pen STAR G640`

SnowLire, 2021-07-04:
Uninstall the `-git` version and use the normal `opentabletdriver` from the AUR. The `-git` version builds from master which is rather unstable at the moment.

BramboraSK, 2021-07-04:
Yeah, I tried that, but it crashes on Connecting to OpenTabletDriver daemon screen.

BramboraSK, 2021-07-04:
I tried running it from terminal and I got this error:
```[bramborask@BramboraPC OpenTabletDriver]$ dotnet ./OpenTabletDriver.UX.Gtk.dll
Exception in Gtk# callback delegate
  Note: Applications can use GLib.ExceptionManager.UnhandledException to handle the exception.
StreamJsonRpc.RemoteMethodNotFoundException: No method by the name 'GetTablet' is found.
   at StreamJsonRpc.JsonRpc.InvokeCoreAsync[TResult](RequestId id, String targetName, IReadOnlyList`1 arguments, IReadOnlyList`1 positionalArgumentDeclaredTypes, IReadOnlyDictionary`2 namedArgumentDeclaredTypes, CancellationToken cancellationToken, Boolean isParameterObject)
   at OpenTabletDriver.UX.MainForm.InitializeAsync() in /home/bramborask/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.5.3.2/OpenTabletDriver.UX/MainForm.cs:line 393
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at Eto.GtkSharp.Forms.GtkSynchronizationContext.<>c__DisplayClass0_0.<Post>b__0()
   at GLib.Idle.IdleProxy.Handler()
   at GLib.ExceptionManager.RaiseUnhandledException(Exception e, Boolean is_terminal)
   at GLib.Idle.IdleProxy.Handler()
   at Eto.GtkSharp.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.App.Run(String platform, String[] args) in /home/bramborask/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.5.3.2/OpenTabletDriver.UX/App.cs:line 49
   at OpenTabletDriver.UX.Gtk.Program.Main(String[] args) in /home/bramborask/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.5.3.2/OpenTabletDriver.UX.Gtk/Program.cs:line 10```

coinworld, 2021-07-04:
Reboot your PC or restart the daemon by doing `systemctl --user restart opentabletdriver`

BramboraSK, 2021-07-04:
That helped, but it still doesn't detect my tablet (even after reboot).

BramboraSK, 2021-07-04:
I found an error in console and now it's fixed. Thanks for your time!

closed 2021-07-04 by BramboraSK