OpenTabletDriver/OpenTabletDriver #1240: Driver not working after closing App and then putting PC to sleep.

pandabro42, 2021-07-05:
So yesterday my driver was working fine and everything and now my driver simply doesn't work. the only things I didn't in between the time of the driver working and not working are 1. close the driver (with the X at the top right, not task manager or some shit) 2. put my PC to sleep. And that's it, I got up today and now the driver just doesn't want to work lol.

Random Things of Note:
- The Driver detects my tablet (CTL-472)
- I'm (pretty sure) I've uninstalled all other drivers
- I've got the driver in Absolute Mode
- I've reset all settings to default and redone everything
- The Driver is set to the correct display
- the driver (Probably) 'broke' after I put my PC to sleep
- I've tried to reset my whole PC (did nothing)
- the app itself loads fine

I'll add all the Console shit from the window I've got open at the time of writing Below:

[Normal:Display]	Detected displays: Virtual Display 0 (1920x1080@0,0), Display 1 (1920x1080@0,0)
[Normal:Detect]	Configuration directory: C:\Users\aagui\Desktop\Files\fucking rename god damn it\Configurations
[Normal:Detect]	Searching for tablet 'Gaomon S620'
[Normal:Detect]	Searching for tablet 'Huion H420'
[Normal:Detect]	Searching for tablet 'Huion HS64'
[Normal:Detect]	Searching for tablet 'VEIKK S640'
[Normal:Detect]	Searching for tablet 'Wacom CTE-440'
[Normal:Detect]	Searching for tablet 'Wacom CTH-470'
[Normal:Detect]	Searching for tablet 'Wacom CTH-480'
[Normal:Detect]	Searching for tablet 'Wacom CTH-490'
[Normal:Detect]	Searching for tablet 'Wacom CTH-670'
[Normal:Detect]	Searching for tablet 'Wacom CTH-680'
[Normal:Detect]	Searching for tablet 'Wacom CTH-690'
[Normal:Detect]	Searching for tablet 'Wacom CTL-4100 Bluetooth'
[Normal:Detect]	Searching for tablet 'Wacom CTL-4100'
[Normal:Detect]	Searching for tablet 'Wacom CTL-470'
[Normal:Detect]	Searching for tablet 'Wacom CTL-471'
[Normal:Detect]	Searching for tablet 'Wacom CTL-472'
[Normal:Detect]	Found device 'CTL-472'.
[Normal:Detect]	Using report parser type 'TabletDriverLib.Tablet.TabletReportParser'.
[Normal:Detect]	Tablet found: 'Wacom CTL-472'
[Normal:Plugins]	Plugin directory: C:\Users\aagui\AppData\Local\OpenTabletDriver\Plugins
**_[Error:System.ArgumentNullException]	Value cannot be null._**  (I don't know too much about code but this might be the issue)
[Normal:Settings]	Output mode: TabletDriverLib.Output.AbsoluteMode
[Normal:Settings]	Filters: 
[Normal:Settings]	Display area: [1920x1080@960,540:0°],
[Normal:Settings]	Tablet area: [23x23@32.675865,22.252756:0°],
[Normal:Settings]	Clipping: Enabled
[Normal:Settings]	Tip Binding: 'MouseBinding: Left'@1%
[Normal:Settings]	Pen Bindings: [0, ], [1, ]
[Normal:Settings]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ]
[Normal:Settings]	Driver is auto-enabled.
[Normal:Settings]	Applied all settings.
[Normal:Settings]	Using theme Light
[Normal:Detect]	Searching for tablet 'Gaomon S620'
[Normal:Detect]	Searching for tablet 'Huion H420'
[Normal:Detect]	Searching for tablet 'Huion HS64'
[Normal:Detect]	Searching for tablet 'VEIKK S640'
[Normal:Detect]	Searching for tablet 'Wacom CTE-440'
[Normal:Detect]	Searching for tablet 'Wacom CTH-470'
[Normal:Detect]	Searching for tablet 'Wacom CTH-480'
[Normal:Detect]	Searching for tablet 'Wacom CTH-490'
[Normal:Detect]	Searching for tablet 'Wacom CTH-670'
[Normal:Detect]	Searching for tablet 'Wacom CTH-680'
[Normal:Detect]	Searching for tablet 'Wacom CTH-690'
[Normal:Detect]	Searching for tablet 'Wacom CTL-4100 Bluetooth'
[Normal:Detect]	Searching for tablet 'Wacom CTL-4100'
[Normal:Detect]	Searching for tablet 'Wacom CTL-470'
[Normal:Detect]	Searching for tablet 'Wacom CTL-471'
[Normal:Detect]	Searching for tablet 'Wacom CTL-472'
[Normal:Detect]	Found device 'CTL-472'.
[Normal:Detect]	Using report parser type 'TabletDriverLib.Tablet.TabletReportParser'.
[Normal:Detect]	Tablet found: 'Wacom CTL-472'

Imma Keep dicking around with the driver and see if anything fixes my issue.

if any more info is need let me know and ill try and get it as quick as I can

Mrcubix, 2021-07-05:
first of all, why would you close the driver, ofc it won't work.
second, if not already done, uninstall wacom drivers.
third, about that argument null exception, it is most likely due to you setting up a plugin wrong or forgetting to fill some fields
fourth, you gotta love these win10 user specific issues, probably related to an issue that's supposed to be fixed when #1112 is merged (some part still need changes apparently)

pandabro42, 2021-07-05:
> first of all, why would you close the driver, ofc it won't work.
> second, if not already done, uninstall Wacom drivers.
> third, about that argument null exception, it is most likely due to you setting up a plugin wrong or forgetting to fill some fields.
> fourth, you gotta love these win10 user specific issues, probably related to an issue that's supposed to be fixed when #1112 is merged (some parts still need changes apparently), tho it could have been fixed if you or the manufacturer (if prebuilt) didn't mess with some of windows' settings

1. well no shit it's not gonna work if the drivers not open I ain't stupid. I only closed the driver before I was about to put my PC asleep which shouldn't cause any problems, and I've always had the driver open when using the tablet
2. unless this shit is just invisible or something I've uninstalled all drivers because bro I've been rooting around in device manager and the add & remove programs shit for the last like ~20 minutes
3. imma be straight with you bro I read that issue and I ain't got a single clue what any of that means

X9VoiD, 2021-07-05:
OTD not working after sleep is a real issue, to be resolved by #1112. (Just press Detect or restart OTD)

Tablet input not working after closing OTD regardless of if you sleep or not is not an issue. (Open OTD again before expecting your cursor to move again if you closed it.)

Plugin folder null reference is not an issue, it's always there on first launch. OTD will create the plugin folder later.

Will close as a duplicate of #552.

pandabro42, 2021-07-05:
> OTD not working after sleep is a real issue, to be resolved by #1112. (Just press Detect or restart OTD)
> 
> ~Tablet input not working after closing OTD regardless of if you sleep or not is not an issue. (Open OTD again before expecting your cursor to move again if you closed it.)~
> 
> Plugin folder null reference is not an issue, it's always there on first launch. OTD will create the plugin folder later.
> 
> ~Will close as a duplicate of #552.~
> 
> EDIT: Send a diagnostics file of _after_ you slept with OTD closed.

how do I do that diagnostics shit Broskie

X9VoiD, 2021-07-05:
OTD -> Help -> Export Diagnostics -> Save it somewhere -> Zip it -> Upload zip here

pandabro42, 2021-07-05:
uhhhhhhhhhh the driver just works now, like this are good all I did was reinstall the driver after login back in. this makes me think that when I first installed the driver it didn't all go smoothly and it caused some shit to fuck up but I guess all's well that ends well

closed 2021-07-05 by pandabro42