OpenTabletDriver/OpenTabletDriver #1241: VIEKK S640 no tablet detected

Aqirito, 2021-07-05:
## Description
I have follow the installation instruction but still the app says: "No tablet detected" 
Bellow is my console log:
[Detect:Info]	Searching for tablet 'Gaomon 1060 Pro'
[Detect:Info]	Searching for tablet 'Gaomon M106K Pro'
[Detect:Info]	Searching for tablet 'Gaomon M106K'
[Detect:Info]	Searching for tablet 'Gaomon M10K'
[Detect:Info]	Searching for tablet 'Gaomon S56K'
[Detect:Info]	Searching for tablet 'Gaomon S620'
[Detect:Info]	Searching for tablet 'Huion 1060 Plus'
[Detect:Info]	Searching for tablet 'Huion G10T'
[Detect:Info]	Searching for tablet 'Huion GC610'
[Detect:Info]	Searching for tablet 'Huion H1060P'
[Detect:Info]	Searching for tablet 'Huion H1161'
[Detect:Info]	Searching for tablet 'Huion H430P'
[Detect:Info]	Searching for tablet 'Huion H610 Pro V2'
[Detect:Info]	Searching for tablet 'Huion H610 Pro'
[Detect:Info]	Searching for tablet 'Huion H640P'
[Detect:Info]	Searching for tablet 'Huion H950P'
[Detect:Info]	Searching for tablet 'Huion HC16'
[Detect:Info]	Searching for tablet 'Huion HS610'
[Detect:Info]	Searching for tablet 'Huion HS611'
[Detect:Info]	Searching for tablet 'Huion HS64'
[Detect:Info]	Searching for tablet 'Huion Kamvas 13'
[Detect:Info]	Searching for tablet 'Huion Kamvas 16'
[Detect:Info]	Searching for tablet 'Huion Kamvas Pro 13'
[Detect:Info]	Searching for tablet 'Huion New 1060 Plus'
[Detect:Info]	Searching for tablet 'Huion osu!tablet'
[Detect:Info]	Searching for tablet 'Huion Q11K V2'
[Detect:Info]	Searching for tablet 'Huion Q11K'
[Detect:Info]	Searching for tablet 'Huion Q620M'
[Detect:Info]	Searching for tablet 'Huion WH1409 V2'
[Detect:Info]	Searching for tablet 'VEIKK A15 Pro'
[Detect:Info]	Searching for tablet 'VEIKK A15'
[Detect:Info]	Searching for tablet 'VEIKK A30'
[Detect:Info]	Searching for tablet 'VEIKK A50'
[Detect:Info]	Searching for tablet 'VEIKK S640'
[Detect:Info]	Searching for tablet 'Wacom CTE-430'
[Detect:Info]	Searching for tablet 'Wacom CTE-440'
[Detect:Info]	Searching for tablet 'Wacom CTE-450'
[Detect:Info]	Searching for tablet 'Wacom CTE-460'
[Detect:Info]	Searching for tablet 'Wacom CTE-640'
[Detect:Info]	Searching for tablet 'Wacom CTE-650'
[Detect:Info]	Searching for tablet 'Wacom CTF-430'
[Detect:Info]	Searching for tablet 'Wacom CTH-460'
[Detect:Info]	Searching for tablet 'Wacom CTH-470'
[Detect:Info]	Searching for tablet 'Wacom CTH-480'
[Detect:Info]	Searching for tablet 'Wacom CTH-490'
[Detect:Info]	Searching for tablet 'Wacom CTH-670'
[Detect:Info]	Searching for tablet 'Wacom CTH-680'
[Detect:Info]	Searching for tablet 'Wacom CTH-690'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100WL'
[Detect:Info]	Searching for tablet 'Wacom CTL-460'
[Detect:Info]	Searching for tablet 'Wacom CTL-470'
[Detect:Info]	Searching for tablet 'Wacom CTL-471'
[Detect:Info]	Searching for tablet 'Wacom CTL-472'
[Detect:Info]	Searching for tablet 'Wacom CTL-480'
[Detect:Info]	Searching for tablet 'Wacom CTL-490'
[Detect:Info]	Searching for tablet 'Wacom CTL-6100WL'
[Detect:Info]	Searching for tablet 'Wacom CTL-671'
[Detect:Info]	Searching for tablet 'Wacom CTL-672'
[Detect:Info]	Searching for tablet 'Wacom CTL-680'
[Detect:Info]	Searching for tablet 'Wacom CTL-690'
[Detect:Info]	Searching for tablet 'Wacom FT-0405-U'
[Detect:Info]	Searching for tablet 'Wacom GD-0405-U'
[Detect:Info]	Searching for tablet 'Wacom GD-0608-U'
[Detect:Info]	Searching for tablet 'Wacom GD-0912-U'
[Detect:Info]	Searching for tablet 'Wacom GD-1212-U'
[Detect:Info]	Searching for tablet 'Wacom GD-1218-U'
[Detect:Info]	Searching for tablet 'Wacom MTE-450'
[Detect:Info]	Searching for tablet 'Wacom PTH-450'
[Detect:Info]	Searching for tablet 'Wacom PTH-451'
[Detect:Info]	Searching for tablet 'Wacom PTH-460'
[Detect:Info]	Searching for tablet 'Wacom PTH-650'
[Detect:Info]	Searching for tablet 'Wacom PTH-651'
[Detect:Info]	Searching for tablet 'Wacom PTH-660 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom PTH-660'
[Detect:Info]	Searching for tablet 'Wacom PTH-850'
[Detect:Info]	Searching for tablet 'Wacom PTH-851'
[Detect:Info]	Searching for tablet 'Wacom PTH-860'
[Detect:Info]	Searching for tablet 'Wacom PTK-1240'
[Detect:Info]	Searching for tablet 'Wacom PTK-440'
[Detect:Info]	Searching for tablet 'Wacom PTK-450'
[Detect:Info]	Searching for tablet 'Wacom PTK-540WL'
[Detect:Info]	Searching for tablet 'Wacom PTK-640'
[Detect:Info]	Searching for tablet 'Wacom PTK-650'
[Detect:Info]	Searching for tablet 'Wacom PTK-840'
[Detect:Info]	Searching for tablet 'Wacom PTZ-1230'
[Detect:Info]	Searching for tablet 'Wacom PTZ-1231W'
[Detect:Info]	Searching for tablet 'Wacom PTZ-430'
[Detect:Info]	Searching for tablet 'Wacom PTZ-431W'
[Detect:Info]	Searching for tablet 'Wacom PTZ-630'
[Detect:Info]	Searching for tablet 'Wacom PTZ-631W'
[Detect:Info]	Searching for tablet 'Wacom PTZ-930'
[Detect:Info]	Searching for tablet 'Wacom XD-0405-U'
[Detect:Info]	Searching for tablet 'Wacom XD-0608-U'
[Detect:Info]	Searching for tablet 'Wacom XD-0912-U'
[Detect:Info]	Searching for tablet 'Wacom XD-1212-U'
[Detect:Info]	Searching for tablet 'Wacom XD-1218-U'
[Detect:Info]	Searching for tablet 'XP-Pen Artist 12 Pro'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01 V2'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 02'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 03'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Mini7'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Pro Medium'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Pro Small'
[Detect:Info]	Searching for tablet 'XP-Pen Star 03'
[Detect:Info]	Searching for tablet 'XP-Pen Star 06'
[Detect:Info]	Searching for tablet 'XP-Pen Star 06C'
[Detect:Info]	Searching for tablet 'XP-Pen Star G430'
[Detect:Info]	Searching for tablet 'XP-Pen Star G430S'
[Detect:Info]	Searching for tablet 'XP-Pen Star G540 Pro'
[Detect:Info]	Searching for tablet 'XP-Pen Star G540'
[Detect:Info]	Searching for tablet 'XP-Pen Star G640 V2'
[Detect:Info]	Searching for tablet 'XP-Pen Star G640'
[Detect:Info]	Searching for tablet 'XP-Pen Star G640S'
[Detect:Info]	Searching for tablet 'XP-Pen Star G960'
[Detect:Info]	Searching for tablet 'XP-Pen Star G960S Plus'
[Detect:Info]	No tablet found.
[Settings:Info]	Output mode: Absolute Mode
[Settings:Info]	Display area: [3840x1080@<1920, 540>:0°],
[Settings:Info]	Tablet area: [0x0@<0, 0>:0°],
[Settings:Info]	Clipping: Enabled
[Settings:Info]	Ignoring reports outside area: Disabled
[Settings:Info]	Tip Binding: [Mouse Button Binding: Left]@1%
[Settings:Info]	Pen Bindings: [0, ], [1, ]
[Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ], [6, ], [7, ]
[Settings:Info]	Driver is auto-enabled.

## System Information:
| Name             | Value |
| ---------------- | ----- |
| Operating System | Win10 Home 64bit
| Software Version | 21H1
| Tablet | VIEKK S640

InfinityGhost, 2021-07-05:
https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Windows-FAQ#my-gaomonhuion-tablet-isnt-detected-but-is-listed-as-supported

(This does need to be updated to include VEIKK in the title, but it applies there too.)

Aqirito, 2021-07-07:
okay edited,

X9VoiD, 2021-07-07:
S640 V2 is already implemented in `master`.

closed 2021-07-07 by X9VoiD