OpenTabletDriver/OpenTabletDriver #1242: Support Request: I'm seeing two cursors on the screen at once (mac)

lukaneek, 2021-07-05:
# Support Request

For some reason whenever I try moving the tablet cursor around the cursor jumps around to two points on my screen making the tablet basically unusable.  I am on a mac.  
## What I've already tried

I've tried deleting my old tablet drivers and finding a solution online but nothing comes up.  
## Diagnostic Information

InfinityGhost, 2021-07-05:
This means that two drivers are running simultaneously. Odds are, the wacom driver and OpenTabletDriver are clashing with one another.

joshuah345, 2021-08-13:
make sure the wacom drivers aren't running.

open Terminal.app
run `pkill WacomTabletDriver` twice

InfinityGhost, 2021-10-30:
Stale support request, assuming it has been resolved. This can be reopened in the case this was a mistake.

closed 2021-10-30 by InfinityGhost