OpenTabletDriver/OpenTabletDriver #1244: Tablet doesn't work in FSX

mrmagic223325BRRRT, 2021-07-06:
## Description
I've encountered the problem of the driver not working in FSX, I really dont want to switch to another driver due to this so I hope  that this can be fixed.
## System Information:
OS: Windows 10 Pro
OS Version: 10.0.19042 Build 19042
Tablet: Xp-pen Star G640 V2
OpenTabletDriver Version: 0.5.3.1

SnowLire, 2021-07-07:
Microsoft Flight Simulator X [requires you to run the program as administrator](https://steamcommunity.com/app/314160/discussions/0/626329820782333837/). As OTD is only a user level driver, it cannot send tablet inputs when an elevated program in focus.

You can solve this issue with VMultiMode Relative Mode (Absolute Input is broken on Vmulti) by following instructions [here](https://github.com/X9VoiD/VoiDPlugins/wiki/VMultiMode).

X9VoiD, 2021-07-07:
Will close as a `wontfix`

closed 2021-07-07 by X9VoiD