OpenTabletDriver/OpenTabletDriver #1245: How can I use dark mode

aestheticqq, 2021-07-06:
I saw on several screenshots that you can use dark mode in this driver, but I looked all over the program and haven't found anything that could relay to dark mode, maybe there is some sort of a plugin that enables it? Anyway, I would like to thank everyone who helped me in solving this little issue

InfinityGhost, 2021-07-06:
Dark mode is a GTK and MacOS feature, supported natively by the operating system. Windows doesn't support dark mode natively, so it has to be hardcoded in which takes way more work than it's worth.

tl;dr dark mode isn't available on Windows

kz-n, 2021-07-10:
( It also works on Linux )

gonX, 2025-12-05:
To clarify, this cannot be fixed in `0.6.x`, but will be solved on `avalonia` branch as we no longer depend on native GUI toolkits to provide GUI features.

gonX, 2026-10-04:
Going to reopen this as it seems that this may be now possible with theme supports in Eto v2.12 onwards: https://github.com/picoe/Eto/releases#release-2.12.0