OpenTabletDriver/OpenTabletDriver #1249: Add support for Parblo Mast10

GoodbyeGhost, 2021-07-07:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    Parblo|
| Product Name    Parblo Mast10|
| Width           301.88mm|
| Height          209.6mm|
| Pressure Levels 2048|
| LPI             5080|

## Diagnostic Information
[otd.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6780226/otd.txt)

##Extra Info
I've been tinkering a bit with this at a surface level to try and get it to work with my tablet, but there's been little luck. I used the parblo a640 configuration request to base the Mast 10 json file below, but there are some settings I don't quite know the function of. It looks like otd is detecting my tablet, but it ends up throwing the deviceioexeption error, so i followed the instructions linked with the error with no visible change. Also, I'm running Artix, if that changes anything.
[Mast10.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6780234/Mast10.txt)

Mrcubix, 2021-07-07:
seem like your distro is messing with the descriptor, i don't think you can get it working in this state

gonX, 2021-07-08:
As the error log states:

> The current user does not have the permissions to open the device stream. Follow the instructions from https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception to resolve this issue.

Start with this and then export new diagnostics

GoodbyeGhost, 2021-07-08:
[otd_updated.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6780872/otd_updated.txt)

I couldn't see a difference in the new diagnostics, but i did get a new error when trying to detect the tablet again:
(OpenTabletDriver.UX.Gtk:2745): GLib-GObject-CRITICAL **: 20:42:24:032: g_ob OBJECT (object)' failed

GoodbyeGhost, 2021-07-08:
Also, it looks like the scope of the instructions mostly applies to those who's supported tablets aren't functioning, as apposed to those who are trying to add unsupported tablets to otd. Maybe adding settings for the Mast 10 in '99-opentabletdriver.rules' could help?

gonX, 2021-07-08:
As you observed, the udev rules only affect existing tablets. You need to add rules matching your tablet.

For your tablet, a complete command would look like this:
```bash
cat <<EOF | sudo tee /etc/udev/rules.d/50-opentabletdriver-additions.rules
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="5543", ATTRS{idProduct}=="004d", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="5543", ATTRS{idProduct}=="004d", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="5543", ATTRS{idProduct}=="004d", ENV{LIBINPUT_IGNORE_DEVICE}="1"
EOF
```

Once the file is added you need to reload udev rules: `udevadm control -R`, and then you should replug the tablet. Then descriptors should show up as intended - this should also clear up whether it's a distro issue or just a permissions issue.

Happy debugging, and thanks for helping in making a new configuration!

GoodbyeGhost, 2021-07-08:
Once I did that, I got the ArgumentOutOfRangeException error, so I followed those instructions, then got this error. Not sure what it means.

[otd_error_current.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6786351/otd_error_current.txt)

X9VoiD, 2021-07-08:
If you're using OTD master then update your configuration to match the new format

GoodbyeGhost, 2021-07-08:
You mean the addition rules I added, or the Mast10 json file? I'm new to this so I'm not exactly sure what changes have been made recently.

X9VoiD, 2021-07-08:
Mast10 json configuration file.

GoodbyeGhost, 2021-07-08:
Yep, copied an Artisul tablet config and changed some settings, now the tablet is being detected. I'm gonna test it out a bit see if there are any problems that arise.

[Mast10_updated.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6786592/Mast10_updated.txt)

GoodbyeGhost, 2021-07-08:
Hmmm, now it's throwing out different errors... the most consistent error is code EACCES (failed to initialize virtual tablet).  Judging from the other error, my json probably doesnt have the right settings for my tablet, or it's just a problem with the initialization string setting.

[otd_error_latest.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6786732/otd_error_latest.txt)
[otd_error_latest2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6786733/otd_error_latest2.txt)

adryzz, 2021-07-08:
EACCES is missing permissions for /dev/uinput

GoodbyeGhost, 2021-07-08:
I thought EACCES was an error, are you talking about evdev? I'm not sure how I would go about giving permissions. Does it have something to do with 10-evdev.conf?

X9VoiD, 2021-07-08:
Make sure your udev rule also includes the rule to allow user access to `/dev/uinput`.

GoodbyeGhost, 2021-07-08:
Ah, ok that makes sense. I added the line 'KERNEL=="uinput", SUBSYSTEM=="misc", TAG+="uaccess", OPTIONS+="static_node=uinput' to the top of my 50-opentabletdriver-additions.rules file. This time I get this error. 
[otd_newest.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6787277/otd_newest.txt)

GoodbyeGhost, 2021-07-08:
Actually, I think it's working now, I forgot to replug the tablet. I'm gonna check just to see if there are more errors to worry about

gonX, 2021-07-08:
Without being too sure, I think the rest of your issues are "simply" caused by incorrect tablet configuration. Only moving in the upper left corner indicates `MaxX`/`MaxY` being too high for the tablet.

IIRC the LangID error means that your permissions aren't set correctly

GoodbyeGhost, 2021-07-08:
Yea, you were right about the MaxX/MaxY being too high for the tablet. Right now I'm wondering why the error doesn't proc all the time if it's something like the permissions not being set correctly. I'm going to be doing some testing to see if I can get any more information out of this.

gonX, 2022-11-07:
Next time the error is happening please submit diagnostics (Help -> Export Diagnostics in the OpenTabletDriver GUI)