OpenTabletDriver/OpenTabletDriver #1262: Support Request:

Haawkeyy, 2021-07-13:
# Support Request
 my designated area is not working.

Basically, i installed opentablet and followed the steps that people show on Youtube, chose my area but its not applying when i try to use it. when i try to use it it goes back to full area for some reason and as its my first tablet ever im not sure how to proceed.
## What I've already tried
 tried searching for solutions in youtube but had no success

## Diagnostic Information

Mrcubix, 2021-07-13:
Fill the template by providing a diagnostic file, no support will be provided until filled
[image]

Haawkeyy, 2021-07-13:
this is the diagnostic.
[diagnostic 1.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6811691/diagnostic.1.txt)

Mrcubix, 2021-07-13:
Your tablet isn't plugged in or is plugged in a monitor / shitty USB hub.

Haawkeyy, 2021-07-13:
yep that my bad, try now to check if something has changed at all
[diagnostics 3.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6811873/diagnostics.3.txt)

Mrcubix, 2021-07-13:
See that one program in `add and remove programs` called `Wacom`, yeah, you need to uninstall their driver, should work fine after that if you didn't mess up with configs

Haawkeyy, 2021-07-13:
all working fine right now. thank you for such a fast reply. appreciate it sir.

closed 2021-07-13 by Mrcubix