OpenTabletDriver/OpenTabletDriver #1263: .NET not working for Winds 11?

Juansword, 2021-07-13:
## Description
I've already installed all the .NET 5... as shown below (Especially Desktop Runtime)
[image]
[image]
But it still gives an error when trying both the installer and the portable... as shown below
[image]

I'm not sure if it's an OS thing or if I'm doing something wrong... Can anyone clarify
## System Information:

| My PC             |  |
| ---------------- | ----- |
| Windows 11 Home | Windows Insiders Program
[image]

Mrcubix, 2021-07-13:
This is most likely an OS issue, not something we can fix

X9VoiD, 2021-07-13:
OTD is already confirmed working for other users using Windows 11. If there would be something wrong in your setup then it would be the presence of a runtime when it's already included in the SDK, only install one or the other.

closed 2021-07-13 by X9VoiD