OpenTabletDriver/OpenTabletDriver #1264: Huion Inspiroy H420X supported but not detected

YzooW, 2021-07-13:
## Description

I have a Huion Inspiroy H420X tablet and it is not being detected by the driver even though it is in the supported list. If I'm doing something wrong please let me know.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10 Home |
| Software Version | v0.5.3.3 |
| Tablet           | Huion Inspiroy H420X |
[Huion Inspiroy H420x.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6812416/Huion.Inspiroy.H420x.zip)

gonX, 2021-07-13:
H420X support is only officially in `master` (0.6.x), so you need to add a custom configuration manually:

https://cdn.discordapp.com/attachments/628651971267788800/864643545108840448/H420X.json

Place this file in the `Configurations` folder of OpenTabletDriver

YzooW, 2021-07-14:
Thanks for that

closed 2021-07-14 by YzooW