OpenTabletDriver/OpenTabletDriver #1270: Huion H430P extremely poor performance (?)

Pebor, 2021-07-17:
## Description

When I use the OpenTabletDriver, my mouse cursor lags behind a lot, sometimes it feels okay, but after few seconds of use starts getting laggier and laggier until not working for several seconds, If I wait a sec and then try using it again, the cycle repeats -> starts to lag harder and harder
I am using Nvidia gpu and have already disabled Force Composite pipeline

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | instantos (linux arch-based) |
| Software Version | compiled git source from the aur package |
| Tablet           | H430P |

InfinityGhost, 2021-07-17:
What are your system specifications? Sounds like a potential garbage collection issue.

Pebor, 2021-07-18:
CPU : Intel i7-3770 
GPU : NVIDIA GeForce GTX 750 Ti 
RAM : 8GB DDR3

DivineGloomy, 2021-08-28:
Did you find a solution to fix the lag cursor?

Pebor, 2021-08-28:
Nope! Still unusable.

gonX, 2021-12-06:
Is this still an issue? If it is, can you upload a diagnostics report (Help -> Export Diagnostics) from OTD please?

Pebor, 2021-12-06:
So, funny thing, I borrowed the tablet to a friend and he experienced it too, after some testing on both windows, arch and ubuntu, we concluded that the tablet is just straight up broken. I never thought of that as an option because it's been just sitting next to me for like 2 years straight.. Sorry! Forgot to take this issue down

closed 2021-12-06 by Pebor