OpenTabletDriver/OpenTabletDriver #1278: Support Request: MacOS Catalina OTD not opening

nloxosu, 2021-07-20:
# Support Request
  Application not opening
 Every time I try to open it, it continues to give me "You can’t open the application “OpenTabletDriver” because it may be damaged or incomplete."

## What I've already tried
 Reinstalling .NET 5, and the driver.

## Diagnostic Information

InfinityGhost, 2021-07-20:
Try [v0.5.3.2](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.5.3.2), this might be a packaging issue.

nloxosu, 2021-07-22:
> Try [v0.5.3.2](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.5.3.2), this might be a packaging issue.

Thanks worked perfectly

InfinityGhost, 2021-07-22:
Reopening this as I need to fix the v0.5.3.3 release asset

InfinityGhost, 2021-07-22:
This should be resolved now.

closed 2021-07-22 by InfinityGhost