OpenTabletDriver/OpenTabletDriver #1279: can't open application

Petron84, 2021-07-20:
## Description
i can't open the application in anyway the application just dont do anything, it doesn't even send an error text
[image]
When i try to open the application via cmd, what else do i have to do after download dotnet from microsoft?
or is it system bug, or anything else?

## System Information:

| Name             | Petron84 |
| ---------------- | ----- |
| Win7 Ultimate |
| latest |
| Veikk S640           |

SnowLire, 2021-07-20:
You need to download .NET 5 Desktop Runtime from Microsoft, install it, and afterwards run the UX.Wpf

Petron84, 2021-07-20:
> You need to download .NET 5 Desktop Runtime from Microsoft, install it, and afterwards run the UX.Wpf
[image]
after downloading .NET 5 Desktop Runtime, i tried to run the UX.Wpf and the same thing occured before with the driver

closed 2021-07-20 by X9VoiD