OpenTabletDriver/OpenTabletDriver #1280: Add support for UGEE M708 v1

LiamTheCookie, 2021-07-20:
# Tablet Information

| Name    UGEE M708 v1        | Value |
| --------------- | ----- |
| Manufacturer  "UGEE"   |
| Product Name    "UGEE M708 v1" |
| Width        (10")   |
| Height      (6")    |
| Pressure Levels "2048 levels" |
| LPI      "5080 LPI"       |

(Pretty much identical at least in looks to the XP-Pen Star03)
(I had to zip the diagnostics export since github apparently doesn't support uploading .json files)

## Diagnostic Information

[diagnostics.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6850642/diagnostics.zip)

InfinityGhost, 2021-07-20:
Are you sure its UGEE rather than UGTABLET? We already have support for the UGTABLET M708, so this potentially is already implemented.

iZute, 2021-07-20:
I only decided to suggest this after OTD wasn't recognizing the tablet
UGEE has this thing were their old tablets don't really like to get recognized by their new drivers and I have a feeling this might be one of the situations related to this
was the m708 added around 2018-2019?

iZute, 2021-07-20:
I am stupid I don't use github ever I'm sorry

iZute, 2021-07-20:
I don't use github I apologize for my stupidity

InfinityGhost, 2021-07-21:
Use [this development build](https://github.com/OpenTabletDriver/OpenTabletDriver/suites/3269657981/artifacts/76130399) and it should work. Do note however that v0.5 settings are incompatible with this development build so it might conflict.

iZute, 2021-07-21:
It still isn't detecting my tablet with the new development build you posted. Deleted old driver before attempting to run as well. Because of this, it probably is a different tablet than UGTABLET. Unless I'm doing something wrong.

InfinityGhost, 2021-07-21:
Can you get me the diagnostics for that build?

iZute, 2021-07-21:
[diagnosticsnew.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6859184/diagnosticsnew.zip)

Mrcubix, 2021-07-21:
Not compatible until USB.NET is implemented

iZute, 2021-07-21:
Ah I see. Well, thanks!

Mrcubix, 2021-07-21:
as a side note, i would like to add that there are as much UGEE M708 variants as there are XP-Pen tablets since the only difference with these tablets are the name, the strings we will dump and the name in the simplified device manager if they feel like it

Mrcubix, 2021-09-09:
Join the [discord server](https://discord.gg/P3GYdUS) when you have time, WinUSB support has been added in pre-release and it will be faster to get it compatible if you join

jamesbt365, 2022-06-18:
This tablet has been supported since.

closed 2022-06-18 by jamesbt365