OpenTabletDriver/OpenTabletDriver #1281: Huion Kamvas 13 tablet undetected

Apichiins, 2021-07-22:
I need help with figuring out how to fix my Kamvas 13 tablet that isn't being detected by the program, I have looked at the faq and uninstalled the drivers and used the cleanup.script, but it still doesn't detect my tablet. Is my tablet perhaps not supported?

Mrcubix, 2021-07-22:
- Visit https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0 then follow instructions there

Mrcubix, 2021-07-22:
Also fill the damn template

```
# Support Request
<!-- What do you need help with? -->
<!-- Be descriptive and use images or video where possible. -->

## What I've already tried
<!-- Reference any wiki pages you've already tried -->

## Diagnostic Information
<!-- Export diagnostics and attach or paste below -->
<!-- In the GUI, click Help -> Export Diagnostics -->
```

InfinityGhost, 2021-10-30:
Stale support request, assuming it has been resolved. This can be reopened in the case this was a mistake.

closed 2021-10-30 by InfinityGhost