OpenTabletDriver/OpenTabletDriver #1282: Add support for XP-PEN Deco Fun L

Mema122, 2021-07-25:
# Tablet Information

| Name            | Value |
| -------- | ----- |
| Manufacturer    | XP-PEN
| Product Name    | XP-PEN Deco Fun L
| Width           | 254mm
| Height          | 159mm
| Pressure Levels | 8.192
| LPI             | 5080

## Diagnostic Information

Mema122, 2021-07-25:
[Diagnostic.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6873877/Diagnostic.zip)

Mema122, 2021-07-25:
I made two configs that detects the tablet ( OpenTabletDriver & Devocub ) but there is a problem, I think Windows itself has a drive and use it instead of a custom driver
[Configs.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6874244/Configs.zip)

Mrcubix, 2021-08-15:
[Deco CT1060.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6988956/Deco.CT1060.zip)
- Extract the content of this zip file in `[OTD Folder]\Configurations\XP-Pen`.
- Start OTD's UX, head to Tablets > Device String Reader, i need you to type the following values:
    - 10429
    - 2310
   and then click on Dump all, it will take a while, then you'll need to save it somewhere and upload it here
   only then we will be able to proceed

PR #1679 (2021-12-13): Update configurations and parsers
- Closes #1101
- Closes #1282
- Closes #1338
- Closes #1401
- Closes #1435
- Closes #1448
- Closes #1500
- Closes #1545
closed 2021-12-13 by InfinityGhost