OpenTabletDriver/OpenTabletDriver #1283: Support Request: OTD area not applying to game and (but not as important) raw input not working.

Yuharaa, 2021-07-25:
# Support Request
OTD area that i've set isn't applying to osu! It had been perfectly fine for a day but after booting my laptop the next day it has had problems since. Another problem, whilst not as important, is that raw input isn't working and leads to my cursor being stuck in the top left corner.

## What I've already tried
I've already tried removing windows ink as that was a possible problem though no success or difference has been made.

## Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.5.3.3",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19042.0",

SnowLire, 2021-07-25:
Disabling both `Raw Input` and `Map absolute raw input to osu! window` will solve your issue. If you had use the in-game sensitivity before, dividing both the `Width` and `Height` by the sens value will give you the correct value to use in OTD.

Yuharaa, 2021-07-25:
Thank you for the fast response, though this has only fixed the raw input not working and not the OTD area problem, as it is still using full area and not the OTD area I have set on OTD. I'm not sure if this is the problem but in the console it  says "Warning - Detect - Failed to set tablet feature 02-02." Thanks again for the quick response.

Mrcubix, 2021-07-25:
uninstall wacom drivers then replug it

Yuharaa, 2021-07-26:
I've tried https://gitmemory.com/issue/OpenTabletDriver/OpenTabletDriver/1202/864387535 and it swaps from full area to my area that I've set repeatedly.

gonX, 2021-07-26:
Yes, the 02-02 error means the tablet is likely already initialized by another driver. Double check that you've uninstalled all other tablet drivers (vmulti and wacom) and then replug

Yuharaa, 2021-07-26:
Mistakes on my behalf, it seems that the Wacom drivers weren't fully uninstalled, thank you for pointing this out.

closed 2021-07-26 by Yuharaa