OpenTabletDriver/OpenTabletDriver #1292: Multi-Binding support not showing up for Windows Build

Gnomey123, 2021-07-28:
## Description

There is only the generic "Press a key or press a mouse button" prompt with a Clear and Cancel button.

[image]

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Win10 21H1 19043.1149
| Software Version | v0.5.3.3
| Tablet           | XP-Pen Deco 03

Mrcubix, 2021-07-28:
modifier + key binding is only available natively in master.
In 0.5.3.3, you'll have to use Script runnier (available in the Plugin Manager) and ahk (You'll have to find out how to make multi key binds by yourself using the documentation)

closed 2021-07-28 by Mrcubix

Gnomey123, 2021-07-29:
thanks for the clarification, got it working!
keep up the great work!