OpenTabletDriver/OpenTabletDriver #1293: Add support for Wacom CTL-4100WLE-S

Alfenstein8, 2021-07-28:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    |Wacom
| Product Name    |Wacom CTL-4100WLE-S
| Width           |200mm
| Height          |160mm
| Pressure Levels |4096
| LPI             |2540

## Diagnostic Information
{
  "App Version": "OpenTabletDriver v0.5.3.3",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "5.11.0.7620",
    "VersionString": "Unix 5.11.0.7620"
  },
  "Environment Variables": {
    "GNOME_DESKTOP_SESSION_ID": "this-is-deprecated",
    "XDG_SESSION_CLASS": "user",
    "USERNAME": "alfenstein",
    "GNOME_SHELL_SESSION_MODE": "pop",
    "DBUS_SESSION_BUS_ADDRESS": "unix:path=/run/user/1000/bus",
    "XDG_RUNTIME_DIR": "/run/user/1000",
    "SSH_AGENT_LAUNCHER": "gnome-keyring",
    "QT_ACCESSIBILITY": "1",
    "PATH": "/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/alfenstein/.dotnet/tools",
    "SESSION_MANAGER": "local/pop-os:@/tmp/.ICE-unix/2047,unix/pop-os:/tmp/.ICE-unix/2047",
    "GIO_LAUNCHED_DESKTOP_FILE_PID": "37667",
    "GJS_DEBUG_OUTPUT": "stderr",
    "LOGNAME": "alfenstein",
    "XDG_SESSION_TYPE": "x11",
    "XDG_MENU_PREFIX": "gnome-",
    "GDMSESSION": "pop",
    "GTK_IM_MODULE": "ibus",
    "GTK_MODULES": "gail:atk-bridge:appmenu-gtk-module",
    "QT_IM_MODULE": "ibus",
    "XDG_CURRENT_DESKTOP": "pop:GNOME",
    "JOURNAL_STREAM": "8:31423",
    "DESKTOP_SESSION": "pop",
    "XAUTHORITY": "/run/user/1000/gdm/Xauthority",
    "GPG_AGENT_INFO": "/run/user/1000/gnupg/S.gpg-agent:0:1",
    "GIO_LAUNCHED_DESKTOP_FILE": "/usr/share/applications/OpenTabletDriver.desktop",
    "USER": "alfenstein",
    "HOME": "/home/alfenstein",
    "LANG": "en_DK.UTF-8",
    "SHELL": "/bin/bash",
    "SHLVL": "1",
    "GJS_DEBUG_TOPICS": "JS ERROR;JS LOG",
    "LIBVIRT_DEFAULT_URI": "qemu:///system",
    "PWD": "/home/alfenstein",
    "MANAGERPID": "1701",
    "_": "/usr/share/OpenTabletDriver/OpenTabletDriver.UX.Gtk",
    "XDG_CONFIG_DIRS": "/etc/xdg/xdg-pop:/etc/xdg",
    "WINDOWPATH": "2",
    "DISPLAY": ":0",
    "XDG_SESSION_DESKTOP": "pop",
    "XDG_DATA_DIRS": "/usr/share/pop:/home/alfenstein/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share:/usr/share",
    "XMODIFIERS": "@im=ibus",
    "SSH_AUTH_SOCK": "/run/user/1000/keyring/ssh",
    "INVOCATION_ID": "af5292aebe6c4229a0f34b58099af1af"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-5/1-5:1.1/0003:0B05:18E5.0002/hidraw/hidraw1",
      "VendorID": 2821,
      "ProductID": 6373,
[... 3908 lines cut ...]
      "Time": "2021-07-29T00:21:37.6100655+02:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'Gaomon 1060 Pro'",
      "Level": 1
    },
    {
      "Time": "2021-07-29T00:21:37.6100783+02:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'Gaomon M10K'",
      "Level": 1
    },
    {
      "Time": "2021-07-29T00:21:37.6100911+02:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'Gaomon S56K'",
      "Level": 1
    },
    {
      "Time": "2021-07-29T00:21:37.610104+02:00",
      "Group": "Detect",
      "Message": "No tablet found.",
      "Level": 1
    }
  ]
}

Mrcubix, 2021-07-29:
[image]
https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception

Alfenstein8, 2021-08-25:
> [image]
> https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception

Now it finds it but then disconnects right after
[image]