OpenTabletDriver/OpenTabletDriver #1294: Support Request: Wacom Intuos CTL4100 no desktop input

Lorphex, 2021-07-29:
# Support Request
 OTD is not letting my Wacom Intuos CTL4100 input pen movement to desktop, but it is showing movement in the debugger visualizer.

## What I've already tried
 Restarting PC, restarting OTD, uninstalling/reinstalling OTD, changing stylus nibs.

## Diagnostic Information

[OTDissue.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6897556/OTDissue.zip)

Mrcubix, 2021-07-29:
Plug your tablet in and try again

Lorphex, 2021-07-29:
> 
> 
> Plug your tablet in and try again (export diag once plugged in, cause it wasn't plugged when the diagnostic was exported)

Updated. Sorry been having this issue for a bit and just re-made the diag a few times.

Mrcubix, 2021-07-29:
- Make sure valorant isn't running in background.
- Let me know if you see a warning about admin privileges when starting OTD
- Uninstall wacom drivers

Lorphex, 2021-07-29:
Closing out of Riot Vanguard fixed it... I really don't want that to be the solution since I like to play Osu! in queues and such. Is there any way we could get around that?

Mrcubix, 2021-07-29:
- First of all, complain to riot for making a mobile copy pasta with a shit kernel level anticheat.
- Second, you could use the windows ink plugin has well as it's [requirement](https://github.com/X9VoiD/vmulti-bin/releases/latest) and it's absolute mode, tho it will come with increase latency.
- fianlly, don't play valorant.

Lorphex, 2021-07-29:
So I'll just take the fat L for this one I guess
I did submit a ticket about OTD being blocked by Vanguard, since controller players are in the highest ranks anyway so clearly it's not working to block alternative input devices.
I'll try the Windows Ink settings.

Mrcubix, 2021-07-29:
if you have a mouse plugged in, then other devices won't work / properly, it doesn't prevent people from using controllers or hardware cheats, and doesn't even block some software cheats, in short, it's bad

Lorphex, 2021-07-29:
That's good to know actually. I might be able to get away with just turning off my mouse while I play Osu!
Thanks for the help / info.

Mrcubix, 2021-07-29:
But like you saw earlier, vanguard block input from OTD anywhere else.
what i was referring to is the fact you can't have multiple pointers devices connected at the same time without encountering issues

Mrcubix, 2021-08-15:
i'll close this as it has basically been solved

closed 2021-08-15 by Mrcubix