OpenTabletDriver/OpenTabletDriver #1297: Minimize to system tray on close (Win10)

Gnomey123, 2021-07-29:
## Description

It would be really nice to have the GUI be minimized when closed, because right now if you close it does stay on the system tray, but then terminates shortly after.

I'm not that fluent in C# Dotnet, but it may be this line of code causing this:

(BTW this issue exists on the master branched compiled for win10 as time of writting)

https://github.com/OpenTabletDriver/OpenTabletDriver/blob/5d66fef41891a84a24a4dd51e948c7444b2d26b0/OpenTabletDriver.UX/MainForm.cs#L99

SnowLire, 2021-07-29:
You can already do that. When you press the minimise button, it minimises to tray.
Close still terminates the program as usual and Minimise doesn't take up taskbar real estate.

Mrcubix, 2021-07-29:
Just minimize it and it will minimize and hide in the tray?

closed 2021-07-29 by Mrcubix

Gnomey123, 2021-07-29:
I guess so, but closing the GUI interface also terminates the Daemon. So when I'm done configuring, my muscle memory tells me to "save" and "close", causing me to then wonder why my tablet cursor isn't moving